emmebrusa/TSDZ2-Smart-EBike-1 #116: Can NOT compile.

KILLERD0G, 2024-08-20:
[image]
Hi, must be something im doing wrong. I can compile and change things with version update 3 april, but i cant use the latest. I have enclosed a screen shot.... Thank you for your help..

KILLERD0G, 2024-08-21:
Oh come on....... Can someone tell me why this bash exe thing is happening..... Please !!!!

KILLERD0G, 2024-08-22:
Does anyone actually manage or read what people put here ? Trying to do some experiments and work out settings for people on here, but not much help from admins... Hmm... Looks like im off to buy a PowerGoat Bike... Thanks guys.

dzid26, 2024-08-23:
The answer is not obvious and you didnt show any effort of troubleshooting. Who knows what you are doing. 

Try different PC, google your warning message, ask chatgpt

KILLERD0G, 2024-08-23:
Well im sorry if i come over all freaked out and frustrated.... I have used my pc and i have used my Alienware Laptop and it still does it. I have taken SDCC out, installed new SDCC, i have latest updates, everything, nothing works. Now aprils version works just fine. I really am trying to support this amazing project. You guys have made such a difference i personally think its amazing. Anyways, i will just wait for the next official release and hopefully take it from there. I know you might not hear this very often but please keep up the good work your doing.. Thanks... PS - i still want a goat bike :)

dzid26, 2024-08-23:
I am assuming you press _compile and flash_ button in java tool.  

What happens if you double click "compile.bat" in "src" folder instead?

KILLERD0G, 2024-08-23:
[image]
I tried again and i still get a bash warning... Yes i am clicking compile and flash in the java tool , when i click complie in src folder the window flashed very fast but it has done something. Problem is when you click compile in the srcf folder, where is it getting the parameters from  ?

dzid26, 2024-08-24:
Seems like it compiled now though?

Perhaps the warning is induced by the java tool.  

Once you click Compile and flash in java tool, the settings are saved in `src/config.h`. 

You can then use any of the .bat scripts from top folder or `src` folder and they will use last saved settings.
`compile_and_flash_20.bat` from top folder is what is called by java, so you can try using that if you still have issues.

KILLERD0G, 2024-08-24:
Oh right, i get you..... Well it flashed and bike seems to run good so far. My system is apparntly a 36v battery and a 250w motor. Basically looking at something that will ghost peddle for me but i can peddle when i fgeel like it. So far this firmware is a god send. Wondering if this firmware will do the TSZD2-B as well ? Thanks for your input. If you want me to try any settings please let me know, im more than happy to try help. But i warn you - im stupid when it comes to technical stuff, but i like to try.. :)