emmebrusa/TSDZ2-Smart-EBike-1 #121: TSDZ2B with EKD01 Display

rekaXua, 2024-09-28:
Hi,
I have bought a TSDZ2B kit with EKD01 Display. 
This is my first time dealing with OFW on Tongsheng, so my question is, can I use this OFW with this display, or is there any OFW that supports it? 
Thank you!

kiwisibk, 2024-10-05:
I would also be very interested if the EKD01 display works with the Emmebrusa software. Can anyone answer that? Thank you

Cyzorp, 2024-10-11:
I tried, with all displays settings best is for vlcd5, there are few problems when switching on erro30 (disappear after 3 sek) and display showing power on level 0 when you push pedals, but working unexpect good. Here is movie with my settings with english subtitles https://youtu.be/-W12oMnz0K4?si=DASh8hrgfKHepE9C

kiwisibk, 2024-10-11:
Thank you so much for your efforts and this precious tutorial!!! 🔝👍👌😊

Cyzorp, 2024-10-11:
Hi, thanks for the message because I often hesitate whether to even
post videos of what I do on YouTube. Even though ekd01 worked with
osf, I recommend 860/850c displays for tsdz2 (must be in the bafang
version) and with osf for tsdz2. I haven't tested this ekd01 with
tsdz2 osf for a long time, so I don't know how it will perform in the
long term (although I think it should work).

Best regards

kiwisibk, 2024-10-11:
Don't worry, this is development, trial & error, nothing ventured, nothing gained! 😉 GitHub is the right platform for this, nobody demands a guarantee of functionality for the tips here. Thanks again 👍

rekaXua, 2024-10-11:
> I tried, with all displays settings best is for vlcd5, there are few problems when switching on erro30 (disappear after 3 sek) and display showing power on level 0 when you push pedals, but working [...]

Bardzo dziękuję panu, bardzo mi to pomogło :)