emmebrusa/TSDZ2-Smart-EBike-1 #125: Creation of a version of OSF for TSDZ8 motor.

mstrens, 2024-12-11:
OSF for TSDZ2 has been greatly appreciated by many users.
The TSDZ8 motor is a good kit (more powerfull, more resistant) that is now used as alternative to the TSDZ2.
Still the original controller firmware for TSDZ8 is much less performant because it does not even allow the user to configure the level of assistance in each mode (ECO, TOUR, ...).
It would be great if an OSF version for TSDZ8 would exist.

The TSDZ8 controller uses another MCU (but still based on cortex M0)  and so part of the code of OSF has to be rewritten.
I would like to do it but it would be easier if someone that already knows OSF code and logic could participate.
@emmebrusa : would you be interested in such a project or do you know someone that would be?
Is it OK that I largely reuse your code? 
If I succeed in this project, the code would be published on github and would be free of charge.

emmebrusa, 2024-12-13:
Already discussed;
[https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues?q=is%3Aissue++TSDZ8](url)

I have 4 bikes with TSDZ2 and I am happy with them, I have no motivation to try TSDZ8.

The code is open source, so you can use it without problems, just respect the terms of the open source license.

mstrens, 2024-12-13:
OK,
I will try to write some code for tsdz8 keeping the terms of the open source license.