emmebrusa/TSDZ2-Smart-EBike-1 #127: Perhaps a small bug + some questions.

mstrens, 2025-01-11:
I continue trying to implement OSF on TSDZ8 controller.
So, I try to understand OSF code.
I noticed a possible issue in motor.c file and I have a question.

First the possible bug.
There is some assembler code bur also the equivalent c code.
I will refer to the c code because it is easier to follow.

`// calculate motor phase current ADC value and update ui8_g_foc_angle
        if (ui8_g_duty_cycle > 0) {
            ui8_adc_motor_phase_current = (uint16_t)((uint16_t)((uint16_t)ui8_adc_battery_current_filtered << 8)) / ui8_g_duty_cycle;
            if (ui8_foc_flag) {
				ui8_adc_foc_angle_current = (ui8_adc_battery_current_filtered >> 1) + (ui8_adc_motor_phase_current >> 1);
                ui8_foc_flag = (uint16_t)(ui8_adc_foc_angle_current * ui8_foc_angle_multiplier) / 256;
                if (ui8_foc_flag > 13)
                    ui8_foc_flag = 13;
                ui8_foc_angle_accumulated = ui8_foc_angle_accumulated - (ui8_foc_angle_accumulated >> 4) + ui8_foc_flag;
                ui8_g_foc_angle = ui8_foc_angle_accumulated >> 4;
                ui8_foc_flag = 0;
            }`
When ui8_g_duty_cycle is small, it could be that (uint16_t)((uint16_t)((uint16_t)ui8_adc_battery_current_filtered << 8)) / ui8_g_duty_cycle exceeds 255.
So ui8_adc_motor_phase_current should be an uint16_t and not an uint_8.
I understand that the risk is low because when ui8_g_duty_cycle is low, the current is also low but at startup, it could be different.

Second : my question.
I understand the principle that the stator phase must get some more advance when the current/power increases.
Still I do not understand the way (ui8_foc_flag) it is just based on ui8_adc_motor_phase_current,
I presume that ui8_adc_battery_current_filtered is the average current on one PWM period (52 usec).
ui8_adc_motor_phase_current is then the max current in one phase during this cycle (when PWM allows the current to from Vcc to grnd).
Why taking then average of those 2 currents to calculate ui8_adc_foc_angle_current?
How is the right value  for ui8_foc_angle_multiplier found/calculated? 

I noticed that this link uses a much more complex code to calculate the g_foc_angle (in motor.c line 1143).
https://github.com/OpenSourceEBike/TSDZ2-Smart-EBike/blob/master/src/motor.c

Thanks in advance for your comments.

emmebrusa, 2025-02-11:
Sorry @mstreni if ​​I answer you late, I hadn't seen these questions, I didn't receive the notification.

I'll start from the last question.
The algorithm for calculating the FOC angle that you posted is the one used in the OSF versions of @casainho, from which my version and that of @mspider65 are forked.
It's a complicated algorithm, I think that whoever developed it started from this document:
[https://www.sciencedirect.com/science/article/pii/S1876610211045309?ref=pdf_download&fr=RR-2&rr=91033c0a697f74ec](url)

In our first versions we used that, then @mspider65 realized that that algorithm greatly underestimated the value and the motor overheated.
So he changed it to a simplified method, now the FOC angle value is calculated once per electric revolution inside the fast PWM interrupt.
@mspider65's algorithm for the current component, is based on ui8_adc_motor_phase_current and ui8_foc_angle_multiplier.
ui8_foc_angle_multiplier is an experimentally obtained constant, which takes into account the different inductance between 36V/48V motors.
When I tried this algorithm in my builds on the bike I have on the home trainer, I noticed one thing, the calculated FOC angle was not always the optimal one, and it was very different between my two builds (860C and stock display).
By manually changing the FOC angle value in different conditions of use (power and ERPS), I compiled a table with the optimal values ​​to obtain the best efficiency.
I then tried to modify the algorithm until I got a calculated table corresponding to the one detected, from this comes 50% ui8_adc_battery_current_filtered and 50% ui8_adc_motor_phase_current.
To be precise I should have done 40/60, but I did 50/50 to simplify the code, I also decreased FOC flag max value from 15 to 13.

In the stock display version, to get the same performance, I had to unify motor.c, eliminating some superfluous code and moving part of it into the main loop of ebike_app.c, the only difference left inside the fast PWM interrupt is the saving of the %SOC, but only this difference introduces a delay in the time of the actual FOC angle that I had to compensate by changing the value of MOTOR_ROTOR_OFFSET_ANGLE from 4 to 2.

Thanks for the report about ui8_adc_motor_phase_current, I will see what to do to avoid the problem by keeping (uinti8_t)ui8_adc_motor_phase_current.

mstrens, 2025-02-11:
Thanks for the detailled answser and explaining the logic/multiplier is an experimental result.

I then have an issue for the TSDZ8 version.

I do not know if the logic is still valid and what should be the best multiplier.
TSDZ8 has only 4 poles (instead of 8) and I am unsure about the inductance.
When I try to measure the inductance, I get quite different values depending on the rotor position.

I expect that the FOC value is very important to get the best efficiency and to avoid overheating.

I do not have a home trainer to make experiment.

Currently I can just let the motor run on the table without any load. In those tests, calculated FOC angle remains 0.

Note: I noticed some other minor potential issues. I will write a separate message for it.

I also plan to keep you informed of the current progress of my TSDZ8 version and the changes I made. I expect it would be better to post this on the forum and not in the github issues

dzid26, 2025-02-16:
`ld  a, xl`  loads lower byte of the result of the `div x, a` , thus the overflow is avoided.

emmebrusa, 2025-02-16:
In my opinion if x > 0xff, _ui8_adc_motor_phase_current+0 should be set to 0xff.

dzid26, 2025-02-16:
Sorry, yes, taking the lower byte `ld  a, xl` obviously doesn't help. I tripped on that one. 

Now I wonder if this could be a reason for a remaining small delay when taking foot of the pedals - it still doesn't not go instantly to zero - even after removing Schmitt-trigger and pedal torque filters...  

Will see.

mstrens, 2025-02-16:
In my version, I set motor_phase_current as uint16_t. 
I was just busy with some test and I got an issue with ui16_adc_motor_phase_current.
Sometime the motor did not start running when I tried to find the best hall offset. For this kind of tests, I set ui8_g_duty_cycle = 0 as initial value.
The issue is then that if the adc battery current is = 1 and g_duty_cycle is 1 (with the ramp up), then the phase current becomes 256 (1<<8)/1 and g_duty_cycle becomes 0 again.
To solve this issue, I just used this code:
`if (ui8_g_duty_cycle > 10) {
                ui16_adc_motor_phase_current = (uint16_t)((uint16_t)((uint16_t)ui8_adc_battery_current_filtered << 8)) / ui8_g_duty_cycle;
            } else {
                ui16_adc_motor_phase_current = (uint16_t)ui8_adc_battery_current_filtered;
            }`
Still, this is probably not the best solution.
Perhaps I could add a test on ui8_adc_battery_current_filtered and set phase_current = battery_current_filtered when battery_current_filtered is less than XXXX (e.g. 20). There should not be a big risk for the motor when battery current is so low.

emmebrusa, 2025-02-17:
At startup ui8_g_duty_cycle = PWM_DUTY_CYCLE_STARTUP.
However, before making changes I want to observe the value of ui8_adc_motor_phase_current to see if it actually reaches 255.

emmebrusa, 2025-02-18:
Of course it needs to be fixed, I meant that I wanted to see what happens first.
Anyway, since you urged me, I think that #0xff should be replaced with ui8_adc_motor_phase_current_max.
The code written in C should be this:

```
uint16_t ui16_temp = ((uint16_t)((uint16_t)ui8_adc_battery_current_filtered << 8)) / ui8_g_duty_cycle;
ui8_adc_motor_phase_current = (uint8_t)(ui16_temp & 0xff);
if ((ui16_temp >> 8) > 0U)
    ui8_adc_motor_phase_current = ui8_adc_motor_phase_current_max;
```

And the compiler output is this:

```
ld	a, _ui8_adc_battery_current_filtered+0
ld	xh, a
clr	a
ld	xl, a
ld	a, _ui8_g_duty_cycle+0
clrw	y
ld	yl, a
divw	x, y
ld	a, xl
ld	_ui8_adc_motor_phase_current+0, a
clr	a
ld	xl, a
tnzw	x
jreq	00054$
mov	_ui8_adc_motor_phase_current+0, _ui8_adc_motor_phase_current_max+0	
00054$:
```

As soon as I have time I will try.

dzid26, 2025-02-18:
Saturating to `ui8_adc_motor_phase_current_max +1` seems unnecessary.   255 is the only logical choice to me. 😀

mstrens, 2025-02-20:
In my code, motor fails to start because I was using some special test code and I started with a g_duty_cycle = 0.

If you started with a duty_cyle = 30, there is probably no issue to start the motor.

Currently in my code I discarded the check on phase current when duty cycle is less than 10. I presume that this is not a security issue.

dzid26, 2025-02-20:
> I tried it and it works, but I can't find a corresponding code in C that compiles it with the same result.

I interpret that as:
```
uint16_t ui16_temp = ((uint16_t)((uint16_t)ui8_adc_battery_current_filtered << 8)) / ui8_g_duty_cycle;
ui8_adc_motor_phase_current = (uint8_t)(ui16_temp & 0xff);
ui16_temp = ui16_temp & 0xff00;
if (ui16_temp > 0U){
    ui8_adc_motor_phase_current = 0xff;
}
```
I don't think it has to generate exact assembly. I think the whole point of inline assembly is that sdcc was inefficient :)

You can also use my handcrafted version which only adds  3cycles. I had inverted conditional jump earlier. 
```
        clrw x          // ui8_adc_motor_phase_current = (ui8_adc_battery_current_filtered << 8)) / ui8_g_duty_cycle;
        ld  xh, a
        ld  a, _ui8_g_duty_cycle+0
        div x, a
        ld  a, xl           // temporarily oad the lower byte of x into accumulator
        cpw x, #0x00ff      // Check if the quotient in x was exceeding 8 bits
        jrc 00050$          // If Carry is set (x <= 255), jump to normal handling
        ld  a, #0xff        // If Carry is set (x > 255), saturate to 255
    00050$:
        ld  _ui8_adc_motor_phase_current+0, a // store the result
```
Corresponding code could be this:
uint16_t ui16_temp = ((uint16_t)((uint16_t)ui8_adc_battery_current_filtered << 8)) / ui8_g_duty_cycle;
ui8_adc_motor_phase_current = (uint8_t)ui16_temp;
if (ui16_temp > 0xff){
    ui8_adc_motor_phase_current = 0xff;
}

Alternatively, we could avoid low precision region and do what casaihno did:
https://github.com/OpenSourceEBike/TSDZ2-Smart-EBike/blob/9f8476fc2917b8282b320f1749b8a508f5e8d42e/src/motor.c#L1170-L1178

dzid26, 2025-02-24:
> At startup ui8_g_duty_cycle = PWM_DUTY_CYCLE_STARTUP

I forget exact history behind this, but if we believe nothing happens below `PWM_DUTY_CYCLE_STARTUP = 30`, shouldn't we subtract this "offset" when calculating the phase current? It would mean we are underestimating the phase current.