emmebrusa/TSDZ2-Smart-EBike-1 #129: Configuring a 48V Motor to 36V via Emmebrusa Software?

kiwisibk, 2025-02-10:
Hello everyone, 

is it possible to set a 48V motor with Emmebrusa software so that it can be operated with a 36V battery? 

Thank you for any helpful tip !!!

kiwisibk, 2025-02-15:
I would like to use the 48V with the 36V battery for a children's ebike. Will it perform worse than a 36V motor with a 36V battery? Or would the performance be comparable?

emmebrusa, 2025-02-15:
I already answered you, they can't have the same performance.
A 48V motor with a 36V battery, will have 3000rpm instead of 4000rpm, also the maximum power decreases proportionally.

kiwisibk, 2025-02-15:
I understand. Can you tell me just for comparison how much a 36V battery
36V motor then has in terms of RPM? Thank you

emmebrusa, 2025-02-16:
For a 36V motor with a 36V battery, the nominal speed is 4000rpm, pedal cadence 97rpm.
The maximum power is about 600W with a charged battery, while as continuous power it is advisable not to exceed 350W.
For a 48V motor with a 36V battery, the nominal speed is 3000rpm, pedal cadence 72rpm and this is a limit.
The maximum power is about 450W and the continuous power is less than 350W, you should check the temperature of the motor.
It can be used but for non-heavy use.

kiwisibk, 2025-02-16:
Ok thanks again.

So definitely not recommended… understood.