emmebrusa/TSDZ2-Smart-EBike-1 #130: Headlights blink and XH18 display shows very large speed

jussisaily, 2025-02-11:
I have a TSDZ2 with OSF bike (XH18 display) with 6V front and read led lights (Bafang) taken from the speed sensor split wiring. If the lights are off, the XH18 display shows normal speed. With the lights switched on, the lights blink briefly every 10-20 seconds or so and the XH18 display shows very big numbers like 50-80 km/h for a few seconds. This causes the odometer to miscalculate the driven trips.

Has anyone else had this problem? The assist does not change so it does not affect driving experience, only the brief light blink, high speed display for s few seconds and too much on the odometer.

I think the firmware version is from April 2022. I'll do a software update when the weather gets warmer here in Finland :-)

emmebrusa, 2025-02-12:
Probably the power of the lights is more than the controller can support. It was discussed here:
[https://endless-sphere.com/sphere/threads/tsdz2-osf-for-all-displays-vlcd5-vlcd6-xh18-lcd3-860c-850c-sw102.110682/page-109#post-1825530](url)

jussisaily, 2025-12-17:
Last night I was driving in the dark and noticed that blinking does not occur at all with assist levels of 2 and higher. It only occurs on assist level 1 with random time period from couple of seconds to 20 seconds. The display probably shows the battery level percentage during the blink for some seconds.

I still have the old Bafang headlight and rear light connected, although I have new lamps to be connected directly to the battery voltage but haven't had the time to connect them. The old lamps directly controlled from XH18 display are very convenient.

Firmware is still from April 2022 too, probably I should update to the latest too... Let's hope the problem will be fixed then.