emmebrusa/TSDZ2-Smart-EBike-1 #131: minor potential bugs

mstrens, 2025-02-11:
1)
When a frame is received from the display, this code applied
`// get wheel diameter from display
			ui8_oem_wheel_diameter = ui8_rx_buffer[3];
			
			// factor to calculate the value of the data to be displayed
			ui16_display_data_factor = OEM_WHEEL_FACTOR * ui8_oem_wheel_diameter;` 

OEM_WHEEL_FACTOR is quite big (=1435)
I do not know the values for oem_wheel_diameter.
Perhaps there is a risk that ui16_display_data_factor exceeds 16 bits.

2)
in ebike.app, there is:
`ui16_adc_pedal_torque_delta = (uint16_t)((ui8_riding_mode_parameter + ui8_pedal_cadence_RPM) >> 2);
`
I am not sure this is correct when the 2 parameters are more than 128 (sum exceed 255)
Perhaps better to have
ui16_adc_pedal_torque_delta =(( (uint16_t)ui8_riding_mode_parameter +  (uint16_t)ui8_pedal_cadence_RPM) >> 2);

3)
In motor.c, there is a variable ui16_cadence_sensor_ticks_counter_min that is increased in some cases (and afterwards used in an if) but this does not seems consistent 
    it is increased only when there is a valid change for the refrence (so only for 1 of the 4 patterns) 
Should it not be increased for all valid pattern changes.
Here the current code.
`if (ui8_temp == ui8_cadence_calc_ref_state) {
                // ui16_cadence_calc_counter is valid for cadence calculation
                ui16_cadence_sensor_ticks = ui16_cadence_calc_counter;
                ui16_cadence_calc_counter = 0;
                // software based Schmitt trigger to stop motor jitter when at resolution limits
                ui16_cadence_sensor_ticks_counter_min += CADENCE_SENSOR_STANDARD_MODE_SCHMITT_TRIGGER_THRESHOLD;
            }`

emmebrusa, 2025-02-13:
Thanks for the reports.
1 - There are no problems, the wheel diameter is in inches, the maximum value is 32.
2 - It needs to be fixed.
3 - Good observation!
But I don't see any problems, when the pedals are moving ui16_cadence_sensor_ticks_counter_min is always updated.
What I was wondering instead, is why CADENCE_SENSOR_STANDARD_MODE_SCHMITT_TRIGGER_THRESHOLD is added to ui16_cadence_sensor_ticks_counter_min and not to ui16_cadence_ticks_count_min_speed_adj in the main loop.
I didn't find an answer, so I left the code as it was, thinking that if it was done like that, there must be a reason.
I will have to look into it in any case.

mstrens, 2025-02-13:
I noticed a typo fout.
in one version you have a #define
MOTOR_SPEED_FIELD_WEAKEANING_MIN ( for 860c version)
and  in the other
MOTOR_SPEED_FIELD_WEAKENING_MIN (VLCD5)

emmebrusa, 2025-02-15:
MOTOR_CALIBRATION_MODE I didn't even remember what it was for!
I'll have to give it a more explicit and understandable name.
It should be HALL_SENSOR_CALIBRATION_MODE, it's an unused function.
In the mspider65 version the control code for the calibration of the hall sensors is in the Android App. It's a complex and heavy code, unthinkable to run it all on STM8.
In the 860C version I added the code only on the controller side, while on the App side I wanted to try to do it on the display.
It's at the bottom of the list of things to do, after the last one!

dzid26, 2025-02-16:
2.

> `ui16_adc_pedal_torque_delta = (uint16_t)((ui8_riding_mode_parameter + ui8_pedal_cadence_RPM) >> 2); `
> I am not sure this is correct when the 2 parameters are more than 128 (sum exceed 255)

in output assembly `bin/ebike_app.asm` there is:
```
;	ebike_app.c: 912: ui16_adc_pedal_torque_delta = (uint16_t)((ui8_riding_mode_parameter + ui8_pedal_cadence_RPM) >> 2);
	clrw	x
	ld	a, _ui8_riding_mode_parameter+0
	ld	xl, a
	ld	a, _ui8_pedal_cadence_RPM+0
	ld	(0x02, sp), a
	clr	(0x01, sp)
	addw	x, (0x01, sp)
	sraw	x
	sraw	x
```

So `addw` is used, which is 16bit. 

In this case SDCC probably simply follows type promotion rules. https://stackoverflow.com/a/46073296/1531161

I think the most explicit and portable way would be:
`ui16_adc_pedal_torque_delta = ((uint16_t)((uint16_t)ui8_riding_mode_parameter + (uint16_t)ui8_pedal_cadence_RPM) >> 2);`