emmebrusa/TSDZ2-Smart-EBike-1 #133: code for filtering could be improved.

mstrens, 2025-02-20:
At several places, you use the filter() function.

The code is 
`uint32_t ui32_temp_new = (uint32_t) ui16_new_value * (uint32_t)(10U - ui8_alpha);
uint32_t ui32_temp_old = (uint32_t) ui16_old_value * (uint32_t) ui8_alpha;
uint16_t ui16_filtered_value = (uint16_t)((ui32_temp_new + ui32_temp_old + 5U) / 10U);`

Imagine e.g. that:
- old value = 0
- new value = 3
- alpha = 8

at first run, filtered value should be (3*2 + 0 + 5)/10 = 1
a second run with the same new value gives (3*2 + 1*8 + 5)/10 = 1
a third run with the same new value will again give 1

So, if new value remains on 3; filtered value stays on 1.

I expect that the only way to solve this issue is to increase the number of bits of old and new values before calling filter()

dzid26, 2025-02-20:
Full code:
https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/b4ada8b6cad8794e676611a84940858fbd66c945/src/common.c#L78-L95

> new value = 3
> alpha = 8

ui32_temp_new = (uint32_t) ui16_new_value * (uint32_t)(10U - ui8_alpha); 
ui32_temp_new = 3 * (10 - 8) = 6, not 32 so I am confused.

> (32 + 0 + 5)/10 = 1

37 / 10 = 3 not 1, so I am confused.

mstrens, 2025-02-20:
Sorry I made a typo fout
It should be
at first run, filtered value should be (3*2 + 0*8 + 5)/10 = 11/10 =  1
a second run with the same new value gives (3*2 + 1*8 + 5)/10 = 19/ 1 = 1
next runs, with same new value do not increase the filtered value. It remains 1.

dzid26, 2025-02-20:
Can you use `code blocks` for equations (Ctrl+E)?  I think Markdown is replacing `*` with italics and they are illegible.

mstrens, 2025-02-21:
Indeed, it was not a typo fout.
I did not know Markdown was replacing some char inside the text.

commit 531f0fb2 (2025-04-13): test filter function, closes #133
closed 2025-04-13 by emmebrusa