emmebrusa/TSDZ2-Smart-EBike-1 #134: Possible efficiency improvement.

mstrens, 2025-02-20:
Trying to develop a version for TSDZ8 controller, I had to fill parameters like those:
`#define MOTOR_ROTOR_OFFSET_ANGLE  (uint8_t)2
#define PHASE_ROTOR_ANGLE_30  (uint8_t)((uint8_t)21  + MOTOR_ROTOR_OFFSET_ANGLE - (uint8_t)64)
#define PHASE_ROTOR_ANGLE_90  (uint8_t)((uint8_t)64  + MOTOR_ROTOR_OFFSET_ANGLE - (uint8_t)64)
#define PHASE_ROTOR_ANGLE_150 (uint8_t)((uint8_t)107 + MOTOR_ROTOR_OFFSET_ANGLE - (uint8_t)64)
#define PHASE_ROTOR_ANGLE_210 (uint8_t)((uint8_t)149 + MOTOR_ROTOR_OFFSET_ANGLE - (uint8_t)64)
#define PHASE_ROTOR_ANGLE_270 (uint8_t)((uint8_t)192 + MOTOR_ROTOR_OFFSET_ANGLE - (uint8_t)64)
#define PHASE_ROTOR_ANGLE_330 (uint8_t)((uint8_t)235 + MOTOR_ROTOR_OFFSET_ANGLE - (uint8_t)64)

I suspected that the sensors where not perfectly equaly spaced, that there had different sensitivity and hysteresis.

To find the best values, I tried different methods

I first made a version that let the magnetic flux rotate very slowly (with 256 steps per electric rotation) with a fixed duty cycle.
After each step, the progam waits for some milisec and read the hall sensor pattern.
When hall pattern changes, it logs the angle.
This method does not work well. Depending on the duty cycle and the wait times, the results differs quite a lot and the interval between 2 pattern changes is absolutely not the expected 60°.
Futhermore, if I use those values to fill the define, the motor does not run very well.
I expect that due to the friction in the motor, the rotor does not exactly align with the magnetic flux.

So I had to find another method.

In a second trial, I filled some reasonable values in the define and I made a version that changes automatically (but slowly) only one PHASE_ROTOR_ANGLE by an offset from e.g. -8,-7, -6, ..., 6, 7, 8 and I monitor the currrent (with a fixed duty cycle) 
I selected the offset that gives the lowest current.
I repeated this for each of the 6 hall patterns.
I applyied those offsets on the PHASE_ROTOR_ANGLE(s).
Then I tried a new set of test for each of the 6 patterns. I found other offsets that I applied again.
A next set of tests will again give other offsets (hopefully with minor differences).
This process is quite long and results are probably not optimum.

Third, I noticed that the erps is very stable (for a given configuration/duty cycle).
So I decided to take one hall pattern as reference (pattern = 1) and to compare for each other pattern change (2,...6):
- the predefined angle in the table ui8_hall_ref_angles[] 
with
- the real angle based on the number of ticks between pattern 1 and this pattern. I also take care of the number of ticks for a 360° electric rotation.

Then,  I automatically update the table ui8_hall_ref_angles[] with the real angle taking care of a filter and only in some conditions:
- erps must be more than a min
- erps must be unchanged compared to previous rotation
- new real angle may not be to different from the value in the table

This seems to give very good results. Currently I found e.g. angles 24, 65, 106, 151, 194, 234.
The main advantage is that the table updates itself automatically (even ifv duty cycle and rps ).  It allows to take care automatically of differences that could exist between different motors. 

This method does not require a lot of code and part of it could eventually be put in ebike_app.c instead in the interrupt.
For TSDZ8, there is no issue having the code in the interrupt. 
Before those changed, I measured a max time in the first interrupt of 16 usec (so less than 26 usec). So I had some reserve.
Note: iIn my performance test it was the second interrupt that was more critical (max was 22 usec). Did you noticed this also in your TSDZ2 version?

This new method is not yet uploaded on github. but if you are interrested, I can share already my c code.

emmebrusa, 2025-02-24:
If I understand correctly this method is alternative to "Hall sensor calibration" do mspider65.
If you think it can be implemented on TSDZ2 too, send it to me and I'll look into it.

mstrens, 2025-02-24:
You can find the full code on github ( https://github.com/mstrens/OSF ).

As TSDZ8 uses a mcu 32 bits at 32mHz (and timers at 64mHz), I use only C code.

Changes about hall sensors positions are in motor.c.

Before trying to optimise hall sensors positions, I made a few changes in order to simplify code (from my point of view).

E.g. I changed some tables about hall pattern to read them directly with the hall pattern code like this one:
`uint8_t ui8_hall_ref_angles[8] = { // Sequence is 1, 3, 2 , 6, 4, 5; so angle are in theory e.g. 39, 86, 127, 167, 216, 0 (256=360°)
        0,                     // error ; index must be between 1 and 6
        24 , //   test gives angle = 39   for hall pattern 1
        106, //   test gives angle = 127  for hall pattern 2  
        65, //    test gives angle = 86   for hall pattern 3
        194, //   test gives angle = 216  for hall pattern 4 
        234 , //    test gives angle = 0    for hall pattern 5
        151, //   test gives angle = 167  for hall pattern 6  
        0                     // error ; index must be between 1 and 6
};
`
I also added a table to check the validity of all hall pattern sequences. ( It seemed easy to to).
`// pattern for hall sensor is 1,3,2,6,4, 5
// with full use of posif possibilities, this table should be read with expected pattern and so upload in shadow register for the next expected
// when current pattern is 1 and expected = 3 , the sadow register should be prepare for the next transition with current = 3 and exp=6 
// in current version, it is used only to detect if a transition is valid in irq0
const uint8_t expected_pattern_table[8] = {
    3, // 0 should not happen
    3, // after 1 => 3 
    6, // after 2 => 6
    2, // after 3 => 2
    5, // after 4 => 5
    1, // after 5 => 1
    4, // after 6 => 4
    1 // 7 should not happen 
};`

My changes are mainly in the code managing pattern changes.
When there is a valid transition and it concern hall pattern 0X01, then I calculate ui32_angle_per_tick_X16shift (=1/number of ticks for 360°).
This allow to make only one division per 360° and for other changes (or for phase current) I can just use a multiply that is faster.
 
Then when it is a transition with an hall pattern different from 0X01, I calculate the angle based on enlapsed ticks since 0X01 change. I added it to the angle defined for pattern 0X01. In fact it would even be easier if angle for pattern 0x01 should be set on zero. 
I assume that the calculated angle for the 5 patterns is better than the one defined in the reference table if
- erps is high enough (rotor has enough inertia)
- number of ticks between 2 consecutive 360 rotations are equal (=speed is stable)
Then I change the values in the angle reference table using some smoothing logic.

If you think it takes to much time for stm8, it should be possible to save data some table in the irq and then to process them in the loop running at 25 msec .
Per pattern (2...6) a table could contain:
- a flag to say if data are valid or not (speed is stable and high enough)
- number of ticks for 360°
- number of ticks since pattern 0X01

Here this part of the code.

`    // when pattern change
    if ( current_hall_pattern != previous_hall_pattern) {
        if (current_hall_pattern != expected_pattern_table[previous_hall_pattern]){ // new pattern is not the expected one
            ui8_motor_commutation_type = BLOCK_COMMUTATION; // 0x00
            ui8_hall_360_ref_valid = 0;  // reset the indicator saying no error for a 360° electric rotation 
            ui32_angle_per_tick_X16shift = 0; // 0 means unvalid value
            //hall_pattern_error_counter++; // for debuging
        } else {   // valid transition
            if (current_hall_pattern ==  0x01) {  // rotor at 210°
                if (ui8_hall_360_ref_valid) { // check that we have a full rotation without pattern sequence error
                    ui16_hall_counter_total = last_hall_pattern_change_ticks - previous_360_ref_ticks; // save the total number of tick for one electric rotation
                    ui32_angle_per_tick_X16shift = ( 1 << 24) / ui16_hall_counter_total;
                    ui8_motor_commutation_type = SINEWAVE_INTERPOLATION_60_DEGREES; // 0x80 ; it says that we can interpolate because speed is known
                }
                ui8_hall_360_ref_valid = 0x01;
                previous_360_ref_ticks = last_hall_pattern_change_ticks ;    
[... 34 lines cut ...]
            //    interval_counter--;
            } 
            if (current_hall_pattern == 1 ){
                ui16_hall_counter_total_previous = ui16_hall_counter_total; // save previous counter (to check if erps is stable)
            }
            previous_last_hall_pattern_change_ticks = last_hall_pattern_change_ticks; 
            //hall_pattern_valid_counter++;     // count the valid frame for debuging
            //hall_pattern_intervals[current_hall_pattern]  = last_hall_pattern_change_ticks - last_hall_pattern_change_ticks_prev; // save the interval between 2 changes
        }
        previous_hall_pattern = current_hall_pattern; // saved to detect future change
        // set rotor angle based on hall patern
        ui8_motor_phase_absolute_angle = ui8_hall_ref_angles[current_hall_pattern];  
    } else { // no hall patern change
`
If this is not clear enough, just ask.

Note: this part find the best relative positions of the hall sensor patterns  2...6 relative to pattern 1.
[... cut ...]
                                i8_hall_add_angles[current_hall_pattern]; // add the value found during each hall pattern optimisation
    `
To optimise, I change this value to get the lowest current for a given duty cycle.
In a first step I wrote some code that let the motor runs with differents values for global_offset_angle (within a range and with a step of 1).
Finally, I removed this code because with the Jlink device used for flashing, it is easy to use a tool displaying some live data (e.g. current) on the PC and letting me to change directly variables like global_offset_angle and duty cycle.

emmebrusa, 2025-02-25:
I think I'll have to study a lot.