emmebrusa/TSDZ2-Smart-EBike-1 #135: Questions about differences between OSF for VLCD5 and for 860C.

mstrens, 2025-02-24:
After having made a beta version of TSDZ8 OSF for VLCD5, I am trying to make a version for 860C.

I noticed that you made a few days ago an update of TSDZ2 860C version.

I used it as basis to prepare the TSDZ8 version. I just made change to it  when it was required for TSDZ8.

Now I just try to compare my TSDZ8 860C version with my TSDZ8 VLDC version.

Doing this comparison, I noticed that there are some quite huge differences between your VLCD5 and 860C versions.

E.g. in ebike_app.c for function apply_cruise, apply_calibration, apply_throttle (here also some parameters can be defined in VLCD5 configurator but are fixed in 860C), apply_speed_limit, check_system.

I presume there is no reason having most of those differences. Do you plan to compare the versions and reduce the differences. I expect it should be easier for the maintenance in the future.

emmebrusa, 2025-02-25:
So, the latest version for 860C is aligned with the latest version for VLCD5 display, not yet released, but you can find it in the repository as master.
There are some differences that are:
- Torque sensor calibration, with 860C it is in the display, with VLCD5 it is in the controller.
- Throttle, with 860C a virtual throttle is available, with VLCD5 not.
- Cruise, they are two modes that have the same name but different functionality.
It was discussed here:
[https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/83](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/83)
Perhaps with different names the two modes could also coexist, but for now I don't have time to dedicate to this, assuming someone is interested.

emmebrusa, 2025-02-25:
FYI, I have more changes planned.
On the 860C display I am adding new features.
On the stock display, I want to revise the torque sensor calibration procedure to make it easier and more intuitive.
I also need to change the display selection in the configurator, not only by type but by supported functions, to easily accommodate new models. So far there are 18!
I also want to try the gear shift sensor.

dzid26, 2025-02-25:
Is gear shift sensor would be to help you reduce torque quickly (after you added the torque rate limit in your settings)?

I don't find it necessary for shifting if the torque delay is low and I can reduce torque with my legs. 

However I noticed the blue gear can slip due to chain shock after the upshifts. This however can be solved with [blue gear slip mitigation] https://github.com/dzid26/TSDZ2-Smart-EBike/commit/800f9192ce2b945fd97c422be41ffba846dd03fa by comparing motor ticks with cadence ticks. Still WIP, because sometimes it still let the motor overrun. (My blue gear barely hold nowadays)

mstrens, 2025-02-25:
I did not saw that there was an update in vlcd5 master. Sorry for that?
I uploaded now the 2 master branches.

I compared ebike_app.c
FYI, I noticed that;
There are several differences about variables being defined as volatile in one version and not in the other.

In 860C there is this code (missing in the other)
`	
    // reset initialization of Cruise PID controller
    if (ui8_riding_mode != CRUISE_MODE) {
		ui8_cruise_PID_initialize = 1;
	}
`

In // reset control parameters if... (safety), there are minor difference about tests on (missing in 860C)
`  || (ui8_assist_level == OFF)
	  || (ui8_riding_mode_parameter == 0U)
	`

There are several minor differences in casting like here 
`        // calculate torque assist target current
        uint16_t ui16_adc_battery_current_target_torque_assist = (ui16_adc_pedal_torque_delta
                * ui8_torque_assist_factor) / TORQUE_ASSIST_FACTOR_DENOMINATOR;
`
Code for static void apply_speed_limit(void) is different. I presume it could be unified.

I also noticed e.g. a difference about (code from VLCD5 version)
`	// Check if the motor goes alone and with current or duty cycle target = 0 (safety)
	if ((ui16_motor_speed_erps > MOTOR_CHECK_ERPS_THRESHOLD)
		&&((ui8_riding_torque_mode) || (m_configuration_variables.ui8_riding_mode == CADENCE_ASSIST_MODE))
		&& (ui8_adc_battery_current_target == 0U || ui8_duty_cycle_target == 0U)) {
			ui8_motor_check_goes_alone_timer++;
	}
	else {
		ui8_motor_check_goes_alone_timer = 0;
	}
	if (ui8_motor_check_goes_alone_timer > MOTOR_CHECK_TIME_GOES_ALONE_TRESHOLD) {
		ui8_system_state = ERROR_MOTOR_CHECK;
	}
	`

I also see one more line (the last one) in vlcd5
`               ui8_system_state = ERROR_MOTOR_BLOCKED;

                // reset motor blocked counter as the error code is set
                ui8_motor_blocked_counter = 0;
 `

dzid26, 2025-02-25:
I would encourage unifying the other display software here as well - for long term maintainability. 

I wish the same could be done with TSDZ8, @mstrens do you think that would be possible? Or are there too many chnages. Different toolchain shouldnt be a big problem, I think.  

Additionally,  if I may advise, @emmebrusa:
- please make smaller commits !!!!
  - one commit per feature or a bugfix, etc
  - learn how to stage individual files and even lines - in VScode it is very easy - I use it for commiting even if I dont use it for editing
- for testing new features it's better to use branches and commit your experiments
  - it's ok if commits are messy there. You can use pull requests (in your own repo) to document, squash and merge the changes in a nice single commit
- use github Release page for changelog (it's automatic!) and for uploading PDFs
  - people will be able to download each release from release page and other binaries like PDFs
  -  dont store PDFs in the repo, only `odt` source needs to be version controlled 
  - dont "version" `odt` files

mstrens, 2025-02-25:
For maintainability, I try to keep the TSDZ8 versions as close as possible from the TSDZ2 versions.

The 860C version is currently quite different from the VLCD5 because:
- part of the logic is in 860C display
- VLCD5 version use a lot of conditional compilation (#if ).

I expect it should be possible to increase the common part of code but it is up to mbrusa to decide. I will follow his choice.

emmebrusa, 2025-02-27:
I checked, there are some differences to fix, they are those introduced in the VLCD5 version with the latest merge (pull requests by @dzid26), they need to be updated in the 860C version, but they do not affect the operation.

> In 860C there is this code (missing in the other) ` // reset initialization of Cruise PID controller if (ui8_riding_mode != CRUISE_MODE) { ui8_cruise_PID_initialize = 1; } `
> 
In the VLCD5 version the initialization of the PID parameters is in apply_cruise()

> In // reset control parameters if... (safety), there are minor difference about tests on (missing in 860C) ` || (ui8_assist_level == OFF) || (ui8_riding_mode_parameter == 0U)`
> 
n 860C the level management is in the display, in ebike_app.c there is a flag, it is only relevant if (ui8_assist_level == OFF).

> There are several minor differences in casting like here ` // calculate torque assist target current uint16_t ui16_adc_battery_current_target_torque_assist = (ui16_adc_pedal_torque_delta * [...]
See introduction.

> I also noticed e.g. a difference about (code from VLCD5 version) ` // Check if the motor goes alone and with current or duty cycle target = 0 (safety) if ((ui16_motor_speed_erps > [...]
"Check if the motor goes alone" is a check present only in the VLCD5 version.
In the old versions it happened that the motor worked alone without control, they called it ghost pedaling.
I think the problem is overcome, but I preferred to still keep the error check.
In the 860C version it is not there because it has never happened.
I could add it but the error byte is full (you saw that on 860C the errors are managed by bits), I would have to add another byte. I will evaluate if it is the case.

> I also see one more line (the last one) in vlcd5 ` ui8_system_state = ERROR_MOTOR_BLOCKED;
> ```
>             // reset motor blocked counter as the error code is set
>             ui8_motor_blocked_counter = 0;
> ```
> 
Forgot to delete it, it is not necessary.

Another important difference.
With VLCD5 the settings made in the configurator become definitions in config.c, they are used partly as constants and partly to initialize variables in ebike_app_init()

With 860C there is no config.c, the settings received from the display are all variables.
There is not even ebike_app_init(),
the processing of the received data (corresponding to the content of ebike_app_init() is done in communications_process_packages(uint8_t ui8_frame_type).

In short, the differences between the two versions are many.
I will evaluate whether it is possible to reduce them, but I have no intention of unifying them, for me understanding the code would be more difficult.
I have already made my choice when I decided to keep the two versions separate.
I say this from experience, when I adopted the 860C display, Casainho had scolded me because I had not maintained compatibility with his version.
And yet I had tried, but how much time wasted in vain!

mstrens, 2025-02-27:
Indeed there are many differences and having only one version does not seems easy.
Still it seems me that it should be possible in ebike_app_init() to copy ALL data being generated by the configurator to variables and to avoid all conditional compilation (replacing #if by normal C "if" based on those variables).
Then the 2 versions would be already a lot closier.

This is what I did (mainly because I wanted that the user did not have to install toolchain tools and to perform compilation).
It allows me to provide just an HEX file for the firmware.
The drawback is that I had to make another version of the configurator to generate the HEX file with the config parameters. As I did not know Java, I did it in XLS but someone knowing java could easily adapt the current java configurator to generate a HEX file instead of a TXT.