emmebrusa/TSDZ2-Smart-EBike-1 #136: Perhaps another way to manage the options for vlcd5 (and similar) display.

mstrens, 2025-02-24:
Currently, with the display keyboard, it is mainly possible to activate/deactivate some options (e.g. the assist mode).

Changing parameters requires to use the configurator and to re-flash.

For sure, the limitted capabilities of VLCD5 display does not allow to display/change all the parameters.

Still, I ask myself, if it would not be possible to let the user select a "profile" with the display keyboard (instead of selecting the assist mode).

With "profile" I mean the full set of parameters defined within the configurator (still one profile would correspond only to one assist mode; the default one at startup). 

If the user want to use all the current assist modes, he could create a profile per assist mode.
Still I expect that most users use only one or two assist mode.
So the user could have e.g. several profiles for the same assist mode. It would be easier to compare in real condition the impact of  some parameters.
It would also e.g. be possible to switch from a 36V to a 48V battery without having to reflash.
This seems me to offer more flexibility.

This suppose that:
- there would be only one version of the compiled firmware (so replacing nearly all #if/#endif by C "if {}"). It is already the way the 860c version is made (even if some parameters are only managed inside the display and not the controller).
- the stm8 has enough flash to store a version containing all options.
- the "profiles" are strored in flash memory (e.g. in the upper part) and can be uploaded separately from the firmware itself.

I understand that this is a quite big change.

What do you think about this proposal?

dzid26, 2025-02-24:
The mental load would be huge and you would need to carry a lookup table to remember what is in the profiles. 
The battery voltage switch is the most compelling use case. (You could almost also automated for yourself based on voltages). Another idea would be to repurpose display speed limit setting for this...  Anyway, I would never use mismatched battery as the max cadence and power would suffer greatly.

However, I was thinking of something similar, but for level selection. - In practise, I use only one assist mode and I only could have a few options switchable.

Given the limitation of vlcd5, my simple use case would be something like:
- Level1: level1 assist
- Level2: Level2 assist
- Level3: Level3 + no speed limit (aka offroad) + field weakening
- Level4: Level4 + no speed limit (aka offroad) + field weakening + enable throttle

Which balances safety and economy.

I find a lot of options in the configurator to be gimmicks and having ability to switch between all of them would add to confusion during the ride, so personally don't care about that. :)

emmebrusa, 2025-02-25:
I think most users use only one mode. I only use Power.
Torque, Cadence, Cruise, I consider them emergency modes, to get home in case of failure.
For this reason it is necessary to be able to change modes on the display.
In my opinion, profiles with many parameters would create confusion.
Keep in mind that the assistance levels are already for each mode and there are other specific parameters for each.
If you want, with "Torque sensor adv" enabled/disabled there is the possibility, limited to the assistance levels, to have two profiles, but only in Power mode.
It could easily be extended to the other modes.
I wrote about it on the Italian forum, it did not attract interest.
Another function that can be considered a profile is Street/Offroad, evaluating which other parameters can be useful, for example in "Street mode" I will add "Motor deceleration".

I had considered unifying the versions, but I came to the conclusion that the disadvantages would be more than the advantages.