emmebrusa/TSDZ2-Smart-EBike-1 #137: Probably better to move some lines of codes.

mstrens, 2025-02-26:
In ebike_app.c, there is some code in the uart_receive_package() function.
`	// current limit with power limit
	ui8_adc_battery_current_max_temp_2 = (uint8_t)((uint32_t)(ui32_adc_battery_power_max_x1000_array[m_configuration_variables.ui8_street_mode_enabled]
		/ ui16_battery_voltage_filtered_x1000));
	
	// set max battery current
	ui8_adc_battery_current_max = ui8_min(ui8_adc_battery_current_max_temp_1, ui8_adc_battery_current_max_temp_2);
`

I think this code that envolve some securirty would be best moved at the end of ebike_control_motor() function because:
- it will be called every 25 msec (instead of 100 msec)
- it will be called even if communication with display is lost.

mstrens, 2025-02-26:
If the check on current is in the UART receive, it is not executed when the controller does not get frame from the display.
It seems me more logic (secure) to do the check without taking care of messages from the display.

dzid26, 2025-02-26:
The uart part just sets the new value for the `ui8_adc_battery_current_max`. 

The limitation will happen outside of UART comm - here, for example:
https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/b4ada8b6cad8794e676611a84940858fbd66c945/src/ebike_app.c#L541-L543

-----
Unless you mean it is important to update the `ui8_adc_battery_current_max` based on latest voltage measurement frequently - to which I fundamentally disagree as explained [above](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/137#issuecomment-2685149241).  :)

mstrens, 2025-02-26:
OK. 
I better understand now.
I also think it makes no sense to adapt the current depending on the actual voltage. It  would be more logic to use the nominal voltage of the battery.

emmebrusa, 2025-02-27:
@mstrens Those lines are in that position because there is no need to update every 25ms, the power limit depends on the choice of Street/Offroad display
and the voltage does not change so quickly.
In the case of no reception there would also be other problems. In fact, you will have seen that after receiving 3 invalid packets ui8_assist_level is set to OFF and the motor stops.

@dzid26 You did not understand the function of those lines of code.
The purpose is to limit the power not the current.
for this the battery voltage is used.
The current is limited by ui8_adc_battery_current_max_temp_1 calculated by "Battery current max (A)" that the user must enter in the configurator based on the characteristics of his battery.
This is the important data to safeguard the battery.
To maintain the power limit, if the voltage decreases the current must increase, but within the limit of ui8_adc_battery_current_max_temp_1.
For example, I set 500W and 17A, the limit with a charged battery is 500W/40V=12.5A, with a discharged battery 500W/29V=17A. Voltage values ​​considering the drop.
With these settings I don't notice any difference in performance between full and empty battery, and that's what I want.

mstrens, 2025-02-27:
OK, I will do the same in TSDZ8 version

dzid26, 2025-02-27:
> The purpose is to limit the power not the current.
> for this the battery voltage is used.

If that's the case, it should be moved to 25ms task, because you want latest voltage readings regardless the UART communication. - In principle. 

>  You did not understand the function of those lines of code.

I understand the lines, but I still didn't hear the purpose of those lines.

However I now remembered one good purpose for this feature! I was going through those thoughts in the past . 
 
This "battery power limit" can be interpreted as _max bike electric power_ (for example for licensing purposes) and it has nothing to do with battery protection as battery doesn't care about power - indeed. 
[image]
The value is surely confusing currently because it is in the battery properties section in java configurator and [rated] Battery Power term is often provided by battery pack sellers. 

So here is my suggestion:
- let's rename Batter Power (W) to Bike Power (W). 
- optionally let's add back Battery Power (W) (for people to enter their rated pack power) - and use constans 3.7V for calculating another current limit

mstrens, 2025-02-27:
I do not think that a battery has a max power.
The battery limit is (I expect) the max current.
In the battery you have usually 3 components:
- cells
- BMS
- fuse
All 3 components are current rated (and not power rate).
If the seller notices a power on the battery, I expect it is based on the the nominal voltage.

I understand that if you want to keep the same power at moter level, it is correct to take care also of the voltage.

dzid26, 2025-02-27:
> I do not think that a battery has a max power.

@mstrens  of course. 

Though I did mention, the manufacturers of battery packs provide their nominal power.  And we can only guess it is a continuous cell current limit x 3.7V and rounded to a nice value. 

My suggestions [above](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/137#issuecomment-2687717594) still hold.

dzid26, 2025-02-27:
>   All 3 components are current rated (and not power rate).

To be clear, I didn't say _power rate_. 
I said "Rated power", aka "nominal power".   But I think you know.  :)

emmebrusa, 2025-02-28:
[image]
Actually this description is misleading, who knows how it started, maybe from the name of the variable, that will have to be changed too. I never noticed.
And to think that on the 860C display, when it was pointed out to me that this parameter was not present in the configuration menu, instinctively I added the "Max power" item to the "Motor" menu.
For now I have changed the description in the configurator, then I will move it to the "Motor settings" section.

dzid26, 2025-03-10:
https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/commit/61f999ecf03d6624fe736bbd1dafa907d248342e  I think this still  will be [misleading](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/152). 

This limit represent Electrical Power, which for the motor it is a sum of Mechanical Power and Power losses (heat), but it is meaningless on its own. 

- For legal reason we want mechanical power output (which corresponds to max bike speed on flat terrain).
- For thermals we want power losses. 

To avoid storing efficiency curves, both could be estimated from [Kbemf](https://github.com/dzid26/TSDZ2-Smart-EBike/blob/c426a77466113f1e994601b30afa2d9b8637c570/src/ebike_app.c#L1575) (or rather its inverse - Torque constant). It's also fairly easy to measure K_bemf for a new motor like TSDZ8 and it is also easiest way to know estimate motor true output torque. 

In the future, I would say we should have 3 power limits:
- Battery rated power max (W) - (based on rated cell voltage 3.7V) which is basically another way to limit battery current
- Motor thermal limit (W) - this could come with some slow low pass filter
- Bike/motor output power (W) - for legal reasons