emmebrusa/TSDZ2-Smart-EBike-1 #138: Probably a #define missing.

mstrens, 2025-02-26:
in ebike_app.c, there is now some checks on OVERCURRENT_DELAY.
Still this is not defined (as far I can see).

Should it not be defined by default with a value > 0 (in order to get a check on overcurrent)?

emmebrusa, 2025-02-27:
OVERCURRENT_DELAY is defined in config.h, I have not updated the file in the repository because it is generated at each compilation.
If you open the configurator and click Compile and Flash, OVERCURRENT_DELAY is at the end of the new config.h file.

dzid26, 2025-02-27:
It should added so that someone can try to compile the defaults without opening Java window.

But maybe it works because somehow CI is not failing https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/actions/runs/13412221225 

I don't know - I will look into the new release properly next week probably. :)