emmebrusa/TSDZ2-Smart-EBike-1 #139: Question about use of throttle

mstrens, 2025-02-27:
A user is testing the TSDZ8 version and he made the following remark:
`
awesome! i noticed today riding home from work the tsdz8 has about 1.5 seconds of lag when you hit the trottle to it actually applying any power... is there any way of changing it so there is less lag in the control system when using the trottle?
`
In principe I reused 100% of the TSDZ2 code. 
Do you notice the same behaviour with OSF for TSDZ2 ?

dzid26, 2025-02-27:
It will depend mainly on the throttle position, shifted gear, and motor inertia.  

In the firmware we **don't** accelerate as fast as possible until cassette synchronizes with wheel speed and if the user holds the throttle lightly it may take a second.

In principle you want to accelerate as fast as possible (max torque) and reduce torque around synchronization speed (`cadence = bicycle speed / gear ratio`) to avoid sudden jerk . But since we don't know for sure what gear the bicycle is in, it would be a bit tricky to do it smoothly especially when user presses throttle lightly and expects little torque and little jerk.

emmebrusa, 2025-02-28:
I don't have a throttle, but on the home trainer I use a virtual throttle and apply_throttle(), like the real throttle, only the input is different, there are no delays.
Years ago, I had been reported the same problem, for this reason I had inserted in the configurator "ADC throttle value min" default = 47 and "ADC throttle value max" default = 176, asking to verify on the display the correspondence of the real values ​​with the default ones.
I had also added "Throttle assist value min" default = 0 and "Throttle assist value max" default = 255, one hypothesis was that, starting from zero, the bike would not move until the driving torque exceeds the resisting torque.
I had therefore asked to try Throttle assist value min" with values ​​greater than zero.
Unfortunately on that occasion I did not receive any answers.
These last two parameters were removed from the configurator due to space requirements in the layout, I think no one has ever used them, but they are still present in the config.h file, instead I have now noticed that they have been lost in ebike_app.c.
When I started testing with the home trainer I increased the value of THROTTLE_DUTY_CYCLE_RAMP_UP_INVERSE_STEP_DEFAULT, the start with the virtual throttle was too abrupt, the chain took a very strong hit.
A long acceleration ramp can give a feeling of delay.
But 1.5 seconds does not seem like much to me, it is not a racing car.