emmebrusa/TSDZ2-Smart-EBike-1 #141: Adding TSDZ8 suport in javaconfigurator.

mstrens, 2025-02-28:
For OSF on TSDZ8, I try to use exactly the same config parameters as your OSF version for VLCD5 (and others similar display).

Still in order to avoid that the user has to install a toolchain for compiling and flashing, I tried to deliver a compiled hex file containing all options allowed by the configurator. It works so.

Still the result of the configurator has to be an hex file instead of a config.h file.

In a first step, I made an XLS sheet to generate the hex file. It works but it does not csupport all checks and all fields foreseen in the javaconfigurator.
It is also difficult more to maintain when you make changes in TSDZ2 version.

I had a look at the code of the javaconfigurator in order to let it generates the HEX file I need. 
I added 2 functions and it seems I succeeded to let it generates an hex file name TSDZ8_config.hex (on top of the other files).

Would you agree to include those 2 functions in your version. The impact that it would generate ome more file that is just discarded by the TSDZ2 users?

I think, this could become easier for maintenance.

Thanks for looking at this proposal.

emmebrusa, 2025-03-01:
Sure, I was suggesting it.
Feel free to send me the edit.

mstrens, 2025-03-01:
Thanks,

To avoid new changes, I still have some checks to do.

Yesterday, I checked that my java configurator version generates an Hex file. It was OK
Today I still have to  check that there are no errors in the HEX file and that TSDZ8 OSF firmware works with it.

In TSDZ8 version, I added some fields:
1) a main version nr : this is used in the firmware to check that the user has really flashed a configuration file and that it is compatible with the OSF firmware version. This version nr is not the firmware version but is specific to the config. It has to be updated only when fields are added/removed/changed in the configurator. There is no reason to change it when bugs are fixed in OSF and/or configurator that does not have an impact on the config parameters
2) a sub version : this field is not really used. I added it just for tracking purpose. I can remove it if you want.
3) some "reserve" : I can remove them. It would be better to add new fields at the end of the current one just like you do (and only when reqired)
4) a global_offset_angle : I use it to fine tune the position of hall sensors. I can compare it to your "#define MOTOR_ROTOR_OFFSET_ANGLE  (uint8_t)2". 
I think it can be useful for the user to modify it to get the best efficiency from the motor.
You have also this comment in motor.c
`
/*---------------------------------------------------------
 NOTE: regarding motor rotor offset

 The motor rotor offset should be as close to 0 as
 possible. You can try to tune with the wheel in the air,
 full throttle and look at the batttery current. Adjust
 for the lowest battery current possible.
 ---------------------------------------------------------*/
`
I suggest to add such a field to the configurator.
There is still a point of attention.
It would be normal that the value would be between e.g. -3 and +3.
There is no issue to generate a #define with a negative value.

This is a little more complex for the HEX file.
When I read the config data in flash, I currently consider that all values are uint16_t (for the float about li-ion cells, I multiply the value by 100 when I generate the HEX file).
To manage negative values, I would have to choose one of this option:
-  consider all values as int16_t (instead of int16_t)  
- add an offset when I generate the Hex file (e.g. +100) and remove it when I read the flash).

It would also be possible to fill the HEX file in such a way that it matches a C structure with different data types but this is more complex (and the stucture must be packed). So I would like to avoid this solution.

The choice I would make could also depend on your preference because you could use the same solution if you decide to remove conditional compilation (#if/#endif) and make only one compiled version using then also a Hex file to flash the user parameters.

Thanks to inform me of your plans.

emmebrusa, 2025-03-01:
The new parameters that I add to the configurator always go at the end of the config.c and ini files, and those that are no longer used I leave them, to maintain compatibility with the previous ini files.

MOTOR_ROTOR_OFFSET_ANGLE is 2 in the VLCD5 version and 4 in the 860C version, even adding the possibility of modifying it, for TSDZ2 it would always be positive.
If you need a range from -3 to +3, can't you add an offset -3 in the C code?
In the configurator I can do from -3 to +3 and in the HEX file from 0 to 6.
What do you think?

mstrens, 2025-03-01:
I do not know the exact range because I have only one motor for testing.
To be safe, I suggest to use a much bigger offset like 100 so it should never be changed.
If you already use 4 for  860C, I suggest to have a range -5/+5 in the configurator and so 95/105 in the Hex file

mstrens, 2025-03-01:
I included my current version of javaconfigurator on github.
It is here https://github.com/mstrens/OSF

It is in tools/java_configurator_source/src.
When you open the java file, you can make a search on mstrens to see where/how I made changes.

In this version, I still have the extra fields I explained here above. I can remove the reserve fields and put the "global offset" at the end. 
We would have to synchronise changes in OSF TSDZ8/TSDZ2 and configurator.

emmebrusa, 2025-03-01:
I saw, they are two separate functions, so no problem.
In the configurator I add globalOffset.
As a description "Hall sensor global offset" is it ok?
Min/max values ​​-5/+5? In the code I see 66.
In the config.c file I add it at the end, if you want to keep the same order, better at the end also in the hex file.
I was thinking of adding in "Motor type" the TSDZ8 choice in order to avoid compiling the TSDZ2 and vice versa.
With a message when finished.
I think I can do it on Tuesday.

mstrens, 2025-03-01:
Of for the names you selected.
Ok for min/max = -5/+5 in the configurator screen; Please add 100 when filled in the HEX file(so it becomes 95/105).
Do not take care of the current 66 value. It is in fact the total offset I currently use in motor.c based on my tests. So it could be considered as the reference  for applying the -5/+5 correction.
When configurator will be modified, I will
- create a #define with a reference 66 (should not be modified by the user)
- calculate the total offset using 66  (=new reference)+ Hall sensor global offset(the new field in HEx file) - 100.

Yes, add the (new) globalOffset at the end of the code and at the end of the HEX. This is the most logic and the easiest for future maintenance.
So you can also remove this part of the code (at the begining of the HEX)
`
mWriter.println(mstrensHexLine(globalOffset)); //04
mWriter.println(mstrensHexLine(reserve_2));  //06
mWriter.println(mstrensHexLine(reserve_3));  //08
mWriter.println(mstrensHexLine(reserve_4));  //0A
mWriter.println(mstrensHexLine(reserve_5));  //0C
mWriter.println(mstrensHexLine(reserve_6));  //0E
mWriter.println(mstrensHexLine(reserve_7));  //10
mWriter.println(mstrensHexLine(reserve_8));  //12
                    
` 
Please keep the first 2 fields (main version and subversion).
It seems me easiest/safiest to have them at the begin to make validity checks in the firmware.

It is indeed a good idea to have a field to identify that it is a configuration for TSDZ2 or TSDZ8.
If you want, you can perhaps just add a third option for the motor type. TSDZ8 exists (as far I know) only in 48V. So we could have e.g.  radio button for
- TSDZ2 36V
- TSDZ2 48V
- TSDZ8 48V.

If you prefer a separate field to make the difference between TSDZ2 and TSDZ8, it is also good for me.

Note: in the version I just published, I have foreseen that the generated hex file is put in a folder named "files_to_flash".
It is also there that I put the compiled fimware.
This seems me clearer for the user to have both files (and only those) in a folder.

I have a question:
as developper I am using vscode and to make those changes to java I installed the microsoft java extension.
I expect it install the java JDK.
I saw on the we that in the left panel, totally at the bottom, there is a tab named Java Projects and there is an icon like |-> (export jar). Could it be that when you use this button, then vscode create a file JavaCodingPack-0.4.2.exe in your download folder that contains every thing that is required to run the configurator and so that the user does not have to install the SDK (at least on windows)?

mstrens, 2025-03-01:
I forget to say that you have to download all files from github (e.g. with the zip file) and to unzip and respect the folder structure.
The jar program uses other files/folders that are on github.

mstrens, 2025-03-02:
While you are working on the configurator, I have here some more proposals.
1) To fine tune the motor, we already decided to add a global offset.
Would it not be good to add also the foc multiplier as input parameter. It seems that you already tried different values and for the TSDZ8 I even do not yet know wich one is good.
2) could you please modify the function I wrote for generating the Hex file in order to have as comment the hexadecimal adress where each line is written in the flash. I already did it for the first lines (e.g. I added 18 in the comment of this line mWriter.println(mstrensHexLine(Integer.parseInt(TF_MOTOR_ACC.getText()))); // 18 MOTOR_ACCELERATION). Still I did not added it for all generated lines and furthermore, it should be changed for the lines where I added it because you should now remove 8 lines at the beginning (global offset and reserves ). If you want I can edit all the comments in my version and send you the code.
3) I do not yey know if it really make sense but for TSDZ8 I prepared some code about temperature. Personnally, I use throttle on the bike for my wife. So I can't read the temperature with the ADC. Still the XMC1302 MCU has an internal temperature sensor. It measures the temperature on the chip and not from the motor but I expect that when the motor temperature increases , then the chip temperature increase too. Perhaps I could use it to protect the motor/controller. In the configurator, it is currently not possible to edit the lower and upper temperature limits when throttle is enabled. Perhaps it could be adapted. This is not critical (I am not sure  that chip temperature is meaningful to protect the motor and TSDZ8 should be less subject to high temperature because it has a better cooling by conception). Do what you like.

emmebrusa, 2025-03-04:
1 - MOTOR_ROTOR_OFFSET_ANGLE and FOC_ANGLE_MULTIPLIER
are delicate parameters, personally I think they should not be entered in the configurator.
I understand that at this stage you have to look for the optimal values, but you can't ask the normal user to do it without providing a clear calibration procedure on how to obtain them, you can't enter them randomly and leave them to the user's feeling.
I wrote that in the VLCD5 version, to improve performance, I had to decrease MOTOR_ROTOR_OFFSET_ANGLE from 4 to 2,
but by feeling 4 seemed better,
I was proven wrong by the result of P_trainer/P_battery.
Reducing the value of MOTOR_ROTOR_OFFSET_ANGLE was a patch, the reason for the difference between the two versions is still to be understood.
It is true that I changed the values ​​of FOC_ANGLE_MULTIPLIER compared to those of mspider65, but only because the algorithm has changed.
Again, I wouldn't put them in the configurator.

2 - Yes, it's better if you change the code you added.

3 - The temperature limits are disabled only to let the user know that they are not used, if you need to change them to do some tests, you can still do it.
Temporarily set "Temperature sensor", change the values ​​and set "Throttle" again.
We'll evaluate what to do later.

mstrens, 2025-03-04:
1 ) In fact MOTOR ROTOR_OFFSET_ANGLE is quite easy to optimize I think. As written in you code, it is just letting the motor run at high speed and no load and looking for the value that provide the lowest current.
If uou prefer not adding them in the configurator, it is not a big issue for me. I will lease some reserve fields at the beginning ot the generated HEX file. So I can , if needed, modify some parameters before flashing without changing the java program.

2) I changed the file but I will make one more check. e.g. I discovered today that I had a bug in the sequence because in your code the assist mode "Hybrid" was not at the expected place (with the other riding mode. Such an situation is not an issue when you generate #define but wel for my logic that is only based on the sequence. I sent you the file to morrow

3) testing temperature is not an issue. The interest was to let the user select the value he wanted. Still, being not able to change the values in the configurator is also not a big issue. I expect that I can hard code the value in TSDZ8 firmware if the concept make sense.

emmebrusa, 2025-03-05:
About MOTOR_ROTOR_OFFSET_ANGLE and FOC_ANGLE_MULTIPLIER I told you only my opinion, but there is also a space problem in the "Motor settings" section.
If you think other fields could be useful, we could do it another way.
I add a "TSDZ8 other setting" Tab
with 10 fields (are they enough?) with description and value.
Read and saved in a new TSDZ8_header.ini file.

Example:
configMainVersion=0
configSubVersion=2
Hall sensor global angle=66
FOC angle multiplier=xx
Available 5=0
Available 6=0
Available 7=0
Available 8=0
Available 9=0
Available 10=0

So if you change something,, you change the content of the file without rebuilding the jar file.
Can it work?

mstrens, 2025-03-05:
In fact, I think there could be 2 types of parameters:
- parameters that should normally be changed only by a developper (e.g. the main and sub version) or by a user having received special instruction (e.g. the foc multiplier).
- parameters that could be changed by the user (e.g. Hall sensor global angle - or better an offset -5/+5; low limit and high limit temperature).

The first type of parameters can best be put in an (ONE) TSDZ8.ini file. It is fine because it allows to easily changing a parameter without having to rebuild the .jar file. Those parameters does not require to be displayed on a special TSDZ8 tab as they can't be updated by a "normal" user. If there is a TSDZ8 tab, it is save that they can't be modified by the user. Developper or "advanced user" could change directly in the .ini.

The second type of parameters should ideally be displayed on the GUI and be editable by the user. When edited by the user, they should be saved. Still here, ideally,  there should be not ONE .ini but one for each saved configuration (like you do for TSDZ2)

Now, to be pragmatic, and for sure at this stage of development, I think that having only one .ini file for the first type of parameters (non editable) is enough. Experience will show if there are really parameters that should be modified by the user (e.g. because different motors require different optimisation parameters.

mstrens, 2025-03-05:
Just by curiosity: I saw that in last version you allow to select between 18 and 19 kHz for PWM.

Is it any reason?

A higher frequency increases probably the loose in the inverter transistors but increases the accuracy of the sinisoidal voltage.
In theory, each PWM frequency requires also a different setup in some parameters about Hall sensor offsets.

If users noticed a difference between 18 and 19 kHz, it is perhaps because they did not change their Hall sensor offsets?

mstrens, 2025-03-05:
I just put on github a new version of TSDZ8 firmware.
I also modified the version I made of java configurator.
I added the comments and checked/changed the sequence of the fields in the HEX file.

I used folowing fields at the beginning
`
        IInteger global_offset_angle = 100 ; // logical value should be -5/+5 but in order to keep a positive value, we add 100
                                                // So 100 in the generated HEX file means 0 in TSDZ8 firmware. The value 100 is substracted in TSDZ8.
            Integer foc_angle_multiplier = 26;  // to be fine tuned; this is just a first basis
            Integer reserve_3 = 0;
             Integer reserve_4 = 0;
             Integer reserve_5 = 0;
             Integer reserve_6 = 0;
             Integer reserve_7 = 0;
             Integer reserve_8 = 0;
 
`

mstrens, 2025-03-05:
I have a question about creatin .jar files.

I created a jar file on my github but some users can't use it. I think there is a version uncompatibility with some java tools/lib.

I have no experience at all with java.
To create the jar file, as I am using vscode, I installed the microsoft java extension for vscode. I expect it installed the java sdk 21 (not sure)
 When I open the explorer in vscode, there is at the bottom a "Java projects" menu that contains the configurator.java.
I select it and click on an icon "export" (|->).
I get a Jar file that I can use (but not all other users).

Is it another way to create a jar file that could be used by every user?

dzid26, 2025-03-05:
1. I installed:
  - https://marketplace.visualstudio.com/items?itemName=redhat.java and 
  - https://marketplace.visualstudio.com/items?itemName=vscjava.vscode-java-dependency and no other crap.  
2. I opened TSDZ2_Configurator.java.  
3. Pressed Ctrl+Shift+P.  Typed `Java. Export Jar...`,  
4. Pressed Enter to select `TSDZ2_Configurator` main class.  
5. The output was jar file.    Unlike [yours](https://github.com/mstrens/OSF/blob/main/OSF_TSDZ8_configurator.jar), this one actually loads a gui for me:  [OSF.zip](https://github.com/user-attachments/files/19097408/OSF.zip)

All that being said, it might be better to use NetBeans IDE if you want to edit the GUI. 

Also, what does it mean tusers cannot open jar file? Are there errors? Maybe they didn't install [Java Runtime](https://www.java.com/en/download/manual.jsp).

mstrens, 2025-03-06:
@dzid26 
This is strange. 
The way you proceed is very similar to mine except that I use an icon instead of typing Export.

On my pc I have no problem to open the GUI. It works as expected when I dubbel click the .jar

Still, if afterwards, I install on the pc the java runtime V8 (folowing the link you provided), then I can't open anymore the .jar.
I get "an java exception has occured" message.

If I open a cmd window and enter the command java -jar OSF_TSDZ8_configurator.jar, then I get the message
Exception in thread "main" java.lang.UnsupportedClassVersionError: TSDZ2_Configurator has been compiled by a more recent version of the Java Runtime (class file version 65.0), this version of the Java Runtime only recognizes class file versions up to 52.0

dzid26, 2025-03-06:
The GUI doesn't appear if I don't select `TSDZ2_Configurator ` after pressing export.
[image]

As for the errors, my guess is that the extension comes with a JDK version that is newer than your system runtime. 

My runtime is fairly old, but I don't have a problem opening it opening it outside of vscode commands. 
```
>> java --version
java 20.0.1 2023-04-18
Java(TM) SE Runtime Environment (build 20.0.1+9-29)
Java HotSpot(TM) 64-Bit Server VM (build 20.0.1+9-29, mixed mode, sharing)
```
While this shows older JavaSE-18:
[image]

I read [here](https://marketplace.visualstudio.com/items?itemName=redhat.java): 

> If you need to compile your projects against a different JDK version, it's recommended you configure the java.configuration.runtimes property in your user settings, eg:

That might be helpful for you

emmebrusa, 2025-03-07:
@mstrens 
Yesterday I made the changes to the configurator, it needs to be checked, but today I can't.
If you want to try in the meantime, you can find the modified files on GitHub in the Java_Configurator_Source/src folder, and also the jar file.
You also need the "other settings" folder that contains the new ini file.
In the "proven settings" folder, the Default_Settings_*.ini files have changed, I also added Default_Settings_TSDZ8, evaluate what values ​​to set.
Let me know if everything is ok.
A question, did you use another editor for the java source file? The tabs of all the lines have changed compared to the one you posted previously.

mstrens, 2025-03-07:
Thanks.
I will test it.

I did not use another editor. I always use vscode editor and I did not change the tab settings.
I only tried to align a litle better the comments (e.g. removing some #define , adding spaces)

mstrens, 2025-03-07:
It seems to be OK.
I did not found a bug.

Just a comment: 
I noticed that you added a field (motor type) to the ini, the HEX and the config files.
In the code for the hex file, you could also add some comments like this:

`
                     if (JCB_MOTOR_TYPE.getSelectedIndex() == TSDZ8) {  // 12A MOTOR_TYPE 1
                        mWriter.println(mstrensHexLine(1));
                     }
                     else {                                            // 12A MOTOR_TYPE 0
                        mWriter.println(mstrensHexLine(0));
                     }

I still have to add this field to my code and modify my readme to explain the changes and to refer to your site.
So I will remove my version of the configurator from my site.

Thanks again.

Note: I noticed that for the header ini file, you added a field name before the value.
This is nice (it helps avoiding mistakes).
Perhaps you could add it also to the other ini files even if it is less useful becasuse they do not have to be edited. It is just a suggestion for you own facility. I do not need it.
`

emmebrusa, 2025-03-08:
Yes, I added a field.
Yesterday I was in a hurry, I forgot to write it.
In the future any addition will be reported to you.

A question, the TSDZ8_header.ini file that you should manage, do I leave it in my repository or delete it?

mstrens, 2025-03-08:
It is good to be informed about future addition but it is not critical as long it concerns new fields being added after the current one.
The hex file is uploaded in flash memory but there is free space enough and TSDZ8 firmware just ignores the bytes after some address. 
Still if you change some existing fields that are used in TSDZ8 firmware, than I would have to adapt the TSDZ8 firmware and the main_version should be updated in the TSDZ8_header.ini. 

As announced, I already totally removed all files/folders used by configurator from my TSDZ8 github site. I explain in the readme section that the user has to use your repository to generate the configuration hex file. So you should best keep the TSDZ8_header.ini file in your repository.

Still, if you prefer, I could also make a copy of all files/folders used by configurator in my repository. The advantage, is that you can then make all the changes you want without having to synchronise with TSDZ8 firmware. I can just adapt TSDZ8 version when I have time. 

I hesitated between the 2 solutions but as I already announced in a previous post that I would use the first one, I did it but it would not be an issue to change for the second solution.

Let me know what you prefer.

Did you also saw this post (3 days ago):
`
Just by curiosity: I saw that in last version you allow to select between 18 and 19 kHz for PWM.

Is it any reason?

A higher frequency increases probably the loose in the inverter transistors but increases the accuracy of the sinisoidal voltage.
In theory, each PWM frequency requires also a different setup in some parameters about Hall sensor offsets.

If users noticed a difference between 18 and 19 kHz, it is perhaps because they did not change their Hall sensor offsets?

Edit : FYI, I think that original TSDZ8 firmware uses 20 kHz (but I used 19 to align with TSDZ2)
`

dzid26, 2025-03-08:
> Still, if you prefer, I could also make a copy of all files/folders used by configurator in my repository. The advantage, is that you can then make all the changes you want without having to synchronise with TSDZ8 firmware. I can just adapt TSDZ8 version when I have time.

I would suggest to point users to Github Releases page that would store a working copy of java configurator (jar) and the main hex file. 

In OSF repo, you could also point to specific commit here using git submodules. Submodules are not great for end users (because require a separate download/cloning) but ideal for devs to exactly know which commit was used and sync to it.

In this scenario it would be even better if Java Configurator had its own repo and being used as submodule by both TSDZ2-Smart-EBike-1 and OSF. But that might be too much to ask @emmebrusa (unless there more benefits to it that I can't think about).

emmebrusa, 2025-03-09:
> 
> I hesitated between the 2 solutions but as I already announced in a previous post that I would use the first one, I did it but it would not be an issue to change for the second solution.
> 
> Let me know what you prefer.
> 
Let's continue with your choice, let's see how it goes.

> 
> Did you also saw this post (3 days ago): ` Just by curiosity: I saw that in last version you allow to select between 18 and 19 kHz for PWM.
> 
> Is it any reason?
> 
At 18 kHz there is better efficiency, at 19 kHz you reach a higher cadence.
In my versions, the possibility of choice has always been there, it had to be changed in the code. I don't think anyone has tried.

A bit of history. The initial OSF project had a PWM frequency of 15.6 kHz which I believe is the same as that of the stock firmware, then @casainho to support a higher cadence increased the frequency to 19 kHz, @mspider65 did the same in his version and I followed.
Later @mspider65 reduced the frequency to 18 kHz, this was his motivation:
_"Switching frequency is a tradeoff between motor speed and efficiency.
The Mosfet of the TSDZ2 controller aren't so fast, and i think 18KHz can be a good compromise to have good efficiency and at the same time to guarantee good operation when you need to use a high cadence."._

I tried it too, I didn't notice any differences, except for the maximum cadence, so I decided to keep 19 kHz, leaving the possibility of setting 18 kHz in the code.
MOTOR_ROTOR_OFFSET_ANGLE is not parameterized because the value is small.
When I started testing with the home trainer, I verified that there is actually better efficiency at 18 kHz. but the choice remained for 19 kHz.
Last month I started developing a new function on 860C.
On that occasion I dug up the performance tables obtained from the tests on the home trainer.
Analyzing the tables I asked myself, why give up better efficiency? Not everyone is interested in going beyond 110 rpm, it was just a matter of adding the choice in the configurator.

A clarification, the tests for the search for the best FOC angle in all conditions of use, both at 18 kHz and 19 kHz, I did only with the 860C version.
Then, having modified the algorithm, I also applied it to the VLCD5 version.
From the comparison I realized that on VLCD5 the efficiency was lower, hence the decision to modify the value of MOTOR_ROTOR_OFFSET_ANGLE, but it was a fallback, I should have redone the tests as for 860C and corrected the algorithm, but it takes too long.

A higher frequency improves the sinusoidal profile if the points also increase, but the points are the same.
With the old FOC algorithm the waveform was very distorted. Now thanks to the code of @mspider65 it is very good, perfect with high currents.
Here's a screenshot to give you an idea:

[image]

It's 19kHz, I don't have an 18kHz one, but I don't remember any difference.

emmebrusa, 2025-03-26:
I updated the Master version adding the EKD01 display.
Also updated the configurator.
Two lines have been added to the hex file.
ENABLE_EKD01 0=DISABLED 1=ENABLED
ASSIST_LEVEL_5_MODE 0=DISABLED 1=BEFORE_ECO 2=AFTER_TURBO