emmebrusa/TSDZ2-Smart-EBike-1 #145: Bash exe error.

KILLERD0G, 2025-03-01:
Hi everyone, can someone please tell me how to get rid of this error. I have installed all the latest software into my pc and laptop.
Every time i try to run the java config tool i keep getting this error. The only time i didnt get it was from the april update 3 version.
Can someone tell me how to fix this please. Thanks so much, keep up the good work, Paul.

[image]

emmebrusa, 2025-03-04:
But does the compilation work?
If you mean this "bash.exe: warning: could not find /tmp, please create!"
It's a harmless warning.
If you want to eliminate it create an empty tmp folder in cygwin_32 and cygwin_64.

KILLERD0G, 2025-03-04:
Hey, i was just logging on to tell you that i could only get it to flash once i updated sdcc compiler to the latest version which is 4.5.0. Thank you for all your help and for answering me. I flashed it tonight, will report back any problems. Keep up the good work.. :)