emmebrusa/TSDZ2-Smart-EBike-1 #148: v20.1C.6 much quieter motor noise

fusenuk, 2025-03-02:
I've just flashed the latest release and my motor now almost has no whine noise when I'm cycling (48v 500w TSDZ2b).

I'd been running v20.1C.2-update-2 since buying the motor in October 2023 and then swapped from the LCD5 to XH18 last month, so had flashed to v20.1C.6-beta-1 and didn't notice any noise difference between those versions but with the release from yesterday I initially thought the motor wasn't kicking in properly when I had the bike up on my stand but then taking the bike out for a few laps it seems to be working as normal (hybrid mode) but there is almost no electrical 'eeeee' whine noise any more. Is this known / intentional?

Either way, thanks for the updates! Good to see others helping with the coding also. 

One of the things I was going to report was that on v20.1C.6-beta-1 when I pressed the lights button on the XH18 to display stats while I was mid-ride, the 'odometer compensation' feature would return the speed to 0mph eventually but then the speed wouldn't work again until I turned the power off and then on again. This doesn't seem to be a problem now though on this latest release.
(I was running my motor flashed with mph but the xh18 display was still set as kmh as I didn't read the manual till tonight on how to switch the speed unit mode on the display itself -  not sure if that may have been involved in why it broke?)

fusenuk, 2025-03-02:
I'll try and do a proper ride this week, as I only did <500m just now to test the flash worked okay, so will have a better grasp on the noise levels then as there were parts of my normal cycle to work where the motor whine noise really stood out, but if it stays this quiet then it's a massive improvement.

fusenuk, 2025-03-02:
Hmm, although looking at the git commits it seems only the configurator changed between the beta 1 release and the final release. So it should have been the same core updates that I already was using on the motor from when I switched displays. I did previously import my original settings file into the beta-1 folder, where as for the final release I used a fresh settings file and just copied manully the battery values over. I did notice some of the other values had new defaults.

dzid26, 2025-03-02:
https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/discussions/101