emmebrusa/TSDZ2-Smart-EBike-1 #150: formatter used?

ruedbi, 2025-03-05:
I have forked the repo for my purposes.
Just in case I want to contribute to the code - which formatter do you use with which settings (as reformatted, but otherwise unchanged code makes reviewing unnecessarily hard) ?
Thanks!

emmebrusa, 2025-03-07:
Do you mean editor?
For C, I use Notepad++ and VS Code.
For Java, I use NetBeans IDE.

ruedbi, 2025-03-07:
within VS code, you can enable any formatter (clang-format, ...)
do you use any of these or do you format manually?

ruedbi, 2025-03-07:
@emmebrusa Regarding netbeans: I tried but failed on the first attempt - not my world ;-)
I plan to test 2 new displays in the coming days: EKD01 and DZ40mini
Maybe you can extend the configurator please? Should be easy for you, I guess.
Then I will provide a patch for the C-code (whatever is to be changed). Thanks!

emmebrusa, 2025-03-08:
> [@emmebrusa](https://github.com/emmebrusa) within VS code, you can enable a formatter (clang-format, ...) do you use any such thing or do you format manually?

I have never worried about this, I use the default settings.
I think the editor adapts to what it finds in the file.
Maybe the tab setting affects empty files.

emmebrusa, 2025-03-08:
> [@emmebrusa](https://github.com/emmebrusa) Regarding netbeans: I tried but failed on the first attempt - not my world ;-) I plan to test 2 new displays in the coming days or how long it takes: EKD01 and DZ40mini Maybe you can extend the configurator please? Should be easy for you, I guess. Then I will provide a patch for the C-code (whatever is to be changed). Thanks!

I plan to change the display management, there are many models, today there are 18!
I have to change the logic, in the C code there will no longer be any reference to the type of display, but only to the functions that differentiate the various models.
The reference to the type of display will only be in the configurator, where the user can easily add new models, choosing the supported functions.

In this perspective, adding references to EKD01 or DZ40mini in the C code would be inappropriate, you can still check which type of current displays fits best.
And if you notice unsupported differences, you can report it and/or modify the code, starting from the most similar current display type.
To make your work easier, as soon as I have time, I will prepare a list of the differences that may exist between the various types.

ruedbi, 2025-03-10:
@emmebrusa like the idea, I was about to propose such a concept already
I have some finding on the ekd01 already and will let you know about them later on
especially the errorcodes are quite different, I found e.g. no error 3 in the range up to ~30, and 3 (input) shows error 14 (on display)
also the 5 assist levels seem to be mapped slightly differently
so the new concept needs to very flexible, I guess
maybe something like a "display analyser mode" would be useful (skip through the assist levels, confirm error-codes, verify batt level scaling) to get the data needed for the configurator - if so, I'd put it into a separate app to not overload the ebike-app ... just my 2 cents
how does Tongsheng solve this? Is there some initial display device detection on startup?