emmebrusa/TSDZ2-Smart-EBike-1 #152: Power in street mode

ruedbi, 2025-03-10:
@emmebrusa 
Is my understanding correct, that if I use e.g. 250W-limited street mode, it's the _peak power_ that is limited to 250W?
Because that's how it looks both from the code and the values on the display.

Background: That's way more restictive than any brand motor does, where the peak power is basically unlimitted, both legally and technically, and the 250W testing procedure is just a joke. So I'm thinking on a modification that limits the _consumed energy_ to an average of e.g. 250Wh instead, using the offroad values as a limiting power value. Just thinking ...

dzid26, 2025-03-10:
Can you add some references regarding legalities around the peak vs average power?  My initial impression is that for legal reasons peak power (hard limit) should be used, not average. - Like there is no comfort or safety or technical reason that would require use of the average.

That said, the power limit calculation maybe could be improved as currently I would say it is too restrictive than it has to be, because it uses electrical power and not mechanical power. 
"Mechanical power = electrical power x efficiency`.  Considering that at low RPM, the efficiency is close to 0%, the difference can be large. In practise we also need to limit the thermal power so it doesn't matter _that_ much.

ruedbi, 2025-03-10:
@emmebrusa 
There's a good summary in the german forum with many links, just get it translated:
https://www.pedelecforum.de/wiki/doku.php?id=rechtliches:start#pedelec_mit_e-motorunterstuetzung_bis_25_km_h

from a legal point of view using averaged energy would IMHO still be more restrictive than what certified motors do right now, but much better than OSF now, and also on the safe side
also no thermal issues should arise for a motor specified for that power
the motor certification is not done at low RPMs, so you can do what you want there and what does not burn the hardware ;-)

dzid26, 2025-03-10:
English for the reference:  https://eur-lex.europa.eu/legal-content/en/TXT/?uri=celex%3A32013R0168 

In any case this governs power limits in order to sell the bike as pedalec. 
I am assuming road rules somehow refer to those certification rules. 

Anyway I find it contradicting.  The max power specified [Article 2 - 2.(h)](https://eur-lex.europa.eu/legal-content/en/TXT/?uri=celex%3A32013R0168#002.002) `(h) pedal cycles with pedal assistance which are equipped with an auxiliary electric motor having a maximum continuous rated power of less than or equal to 250 W`. This is a lower limit, but the measurements of the 30min max power only focuses on ability to deliver rated power without exceeding thermals (i.e. upper limit). So to me not only the peak power is unlimited (as wiki suggests), but also the maximum continuous power. To me the continuous power limit is for consumer protection - so that you "get the power you paid for" - without overheating. On the other hand it is not illegal to undervalue product power (car manufacturers do it all the time for marketing reasons). So I think the regulators fucked up - they should have just used net power for the classification. This is what makes sense for safety. 

Like, even if I try to make sense of it it doesn't make sense:
Say we want to certify and sell bicycles with this software. It seems to me, that all we have to do is say it is rated 250W or less. 
Then in testing facility they will run it at 250W. If the motor temperature in some (arbitrary point) doesn't exceed 20deg, then we are good.  Which makes no sense.  
Please correct me if I am wrong. But seems I am not the only one to come to this conclusion: https://www.pedelecs.co.uk/forum/threads/continuous-rated-output.16233/post-199075

What is clear is next sentence: `where the output of the motor is cut off when the cyclist stops pedalling and is otherwise progressively reduced and finally cut off before the vehicle speed reaches 25 km/h`. That means no [cruise control without pedalling](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/83). 

 
So for the EU, the only power limit is regarding what you can advertise, and not the ebike classification. 
Correct me if I am wrong.

emmebrusa, 2025-03-11:
@ruedbi
There are many pages on the subject, but in my opinion useless in this context.
You have to look at the problem from another point of view, that of responsibility.
To remain within the law, I purchased 5 TSDZ2 36V 250V, the installation of OSF could invalidate the manufacturer's responsibility, but in the event of disputes, I can demonstrate my good faith.
If you purchase a TSDZ2 48V 750W, you have made an illegal choice from the start. The fact that the peak power is lower than that of known brands is irrelevant. What matters is the power declared by the manufacturer.
Do you want me, with OSF, to take responsibility for making your motor legal?
Simple, set "Max speed offroad mode" and "Street speed limit" to 25 km/h, "Motor power max" and "Street power limit" to 250 W.
Doesn't that suit you? Is it limiting?
Set these parameters as you like, with the criteria you prefer, but it is your responsibility.

ruedbi, 2025-03-11:
@emmebrusa 
firstly, I don't want you to do anything
I am very happy with the SW and the fact that it is much more compliant than the origininal SW, even if overly cautios regarding power
**thanks for that!**
I am just thinking about options, and maybe I can help a little as well
secondly, no one can make a 750W rated motor legal, for the reasons you gave
I agree, for a legal setup the motor has to be rated @250W by the manufacturer (BTW: I have the same motors as you, B-version)
given that, I am sure that increasing the peak power as I was considering complies with legal requirements, as the average supplied power does not exceed 250W (or whatever is set), which is less than 250W machanical output power on average in any case
what you gain is higher peak power in changing terrains, while not overloading the motor on long slopes
that's all

dzid26, 2025-03-11:
> as the average supplied power does not exceed 250W

Both peak and average don't matter for the law.
Only manufacturer rated power matters. 

> If you purchase a TSDZ2 48V 750W, you have made an illegal choice from the start

Not necessarily. Online store can probably advertise peak power. I don't know if they are obligated to advertise rated power? 
It would be illegal **for them** if they sold motor with 750W rating and the temperature increased more than 20C.  That means they didn't go through CE certification testing process.  The CE logo carriers a lot of responsibility.

It seems though that all TSDZ2 are rated 250W - even 48V. I couldn't find with google image search anything else than 250W. 

TSDZ2 were rated 250W (it wouldn't make sense for them to rate it higher).
I haven't found >250W plates with Google Image search even for 48V motor.  
[image]
[image]
Or not rated at all.
[image]

So what to do? According to  https://wooshbikes.co.uk/?tsdz2:
> he motor needs to be labelled in accordance with EN15194 to be road legal. It must contain the manufacturing identification and the 205W rating.
The label must be produced and affixed by Tongsheng for them and us to honour warranty and insurance. It must not be tampered with.

For me unfortunately the power text either disappeared  or was not there. 
[image]

The law is poorly written and unnecessarily opened to interpretation and could be used against the riders for no practical reason. Sad bureaucracy really. 

One option is to refresh the sticker https://www.aliexpress.com/i/1005001958014476.html

ruedbi, 2025-03-12:
@dzid26 I think you are right: The 250W legal thing is only determined by the sticker. 
The applied output power can be virtually anything from legal point of view.
Stupid ;-)

Technically of course it should not overheat / destroy the motor ...