emmebrusa/TSDZ2-Smart-EBike-1 #155: Extreme lack of power in emtb

69pjc69, 2025-03-10:
I solely use emtb and have done since it's release to the tsdz2. However with the latest version it seems very underpowered compared to previous versions. There is something not quite right with it I have used the same or similar settings so can rule out any user error. Only time it seems to wake up is when the number is 254. Anything lower even 230 and it has virtually no power. I have also noticed that the girls weakening doesn't behave aswel either, with previous versions I could get 100-115 rpm at the cranks. New version barely goes past 100 no matter which setting I choose, ie 18 or 19. I am still playing around with the various settings but so far I prefer the older versions

ruedbi, 2025-03-10:
just a guess, but with new FW, the data memory should be flashed / cleared as well (data_empty.hex, https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/src/data_empty.hex) - did you do that too?

emmebrusa, 2025-03-11:
@69pjc69
Limited power even when setting 254?
Something is wrong.
Post the ini file that I check.

You will have seen that in the new version the scale has changed, the sensitivity of the eMTB is no longer from 1 to 20, but from 21 to 254, before there were 20 predefined tables, now the sensitivity is calculated by an algorithm, the minimum value 21 corresponds to the previous table 3, the maximum value 254 to table 20.
Now it is also possible to choose the type of progressivity mode. "Torque-based" which is the same as the previous version, or "Power-based" which is new, everyone can choose the one they prefer.

I will start by saying that I do not use eMTB, but after the modification I tried it with the 860C version.
In the stock displays version, the code is the same.
eMTB assist level 4 - TURBO default is 180, I think that with that value you get close to the maximum power.
Anyway, I'll try again as soon as I have a chance.

Regarding Field weakening, it activates when duty cycle PWM% reaches 100%, if for some reason the power is limited, probably PWM% doesn't reach 100%.