emmebrusa/TSDZ2-Smart-EBike-1 #156: Torque sensor : Not an issue for TSDZ2 but wel for TSDZ8

mstrens, 2025-03-11:
The OSF version for TSDZ8 has been tested on a bike by ebikestuff.

He made the following comment:

`
We did 60 km test run. The max amps value recorded by the BMS was 17.5A. I believe it reached 18A on the display when the amps info was displayed. I had issues with reading the amps and changing config from the display because we have 860C display on our test bike. Switching lights is done by holding the + button. It works differently than on VLCD5. We will probably also test VLCD5 and EKD01 to see if those displays work correctly.
The mode was set to hybrid. The display selection was set to 850C. We used the same settings we use on our TSDZ2 bikes. The riding feel is similar. However, as mentioned on Endlesssphere, there seems to be a problem with the torque sensor sensitivity. The responsiveness is great. However, it's so sensitive that it detects the moment when the crank arms are vertical causing a drop in the pressure applied to the torque sensor. The pressure is not constant because of the way pedalling on a bike works. The current implementation seems to be detecting these changes in the pressure applied to the pedals while pedalling. This makes the assistance to be rather "jerky". It feels like the assistance is being applied as "pulses". Stronger one when there's more pressure on the pedals and a weaker one when the pressure drops. OSF for TSDZ2 doesn't have that problem.
Is there a setting to change the torque sensor sensitivity that might be incorrect? Or maybe this requires a better sync with the cadence sensor to detect when there is a changing in the pedalling/cadence and only then lower the assistance.
`
The code in ebike_app_c for TSDZ8 is the same as the latest code for TSDZ2.
Changes were only done to support the TSDZ8 micro processor and to have only one compiled version (containing all options).
I made more changes in motor.c but the logic is the same (PWM 19kHz, sinusoidal voltages, ...).

I do not see why TSDZ8 version would be so sensitive.
One little difference is that the torque PWM uses a pulse of 2.5 usec (instead of 2 usec). The period is the same (20usec). I used a pulse of 2.5 because that is the value used by the original TSDZ8 firmware. 
When there is no load on the pedal, the ADC torque value on my motor is about 160 (so I expect it is normal).
I do not know my max value of ADC torque (motor is not in a bike).
I use the same parameters as TSDZ2 for all accelerations/decelerations.

Do you have any idea to explain the difference between TSDZ2 and TSDZ8?

emmebrusa, 2025-03-12:
The first thing to do is to observe the value of
pedal torque adc delta (number 9 in the list of displayable variables) to see if the value is stable, it must be zero without pushing on the pedals, and gradually increase by pushing.
A procedure to observe a variable on the display continuously is described 
[here](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/151).

There is a parameter that can make the pedaling smoother, decrease the value of "Motor deceleration", try gradually because it increases the motor stop time.
There is also a parameter to change the sensitivity of the torque sensor with little push.
It is "Pedal torque adc angle adjustment", try decreasing it up to -20.

mstrens, 2025-03-12:
Thanks for the suggestions.
Still afterwards I got a comment saying that the issue (jerking) occurs also when assist mode is set on cadence.
As torque sensor is not used in this mode, there must be another reason. Perhaps the foc multiplier(?).

Still about torque sensor, I have 2 small questions.
In the configurator, when box calibrated is not ticked, the Pedal torque ADC step is default on 67 (with no load = 150 and max load = 300). If I tick Calibrated and Estimated, the value becomes 37. Is this normal?

For info, ebykestuff that is testing the TSDZ8 found a adc no load = 160 and a adc max load = 450.
Is this value for max load also common for TSDZ2?

emmebrusa, 2025-03-13:
From the ebikestuff report it seems that the problem is attributable to the torque sensor.
The problem report also in cadence mode, does it always come from ebikestuff?
If it comes from a different source it could be a different problem.
Do you have any sure news of the correct functioning with the throttle?
If it works well with the throttle, the problem does not depend on the management of the motor, foc or other.

Regarding the calibration of the torque sensor, the check on "Calibrated" is used to confirm the correct entry of the values ​​of "Pedal torque adc offset" and "Pedal torque adc max".
The check on "Torque sensor adv" is used to have the advanced mode enabled at startup.
Advanced mode means that the calibration values ​​are used.
It is a double confirmation because if it were only on the display, you could enable the advanced mode with the default values.
If "Calibrated" and "Torque sensor adv" are not both enabled, "Pedal torque adc step adv" and "Pedal torque adc offset" and "Pedal torque adc max" are not used, only "Pedal torque adc step" is used.
This parameter was used, when there was no calibration, to increase or decrease the assistance in Power mode, but it was an improper use and the human power was wrong.
With the calibration it would be useless, I left it for those who were used to using it in this way.
With the check on "Estimated", the value of "Pedal torque adc step" changes from 67 to another value (it is an estimated value, to obtain a real value you should do the calibration with a weight, it is used for the correct calculation of the human power).
Attention, "Estimated" must be enabled only if the real values ​​have been entered in "Pedal torque adc offset" and "Pedal torque adc max".
If "Calibrated" and "Torque sensor adv" are both enabled, "Pedal torque adc step" is not used.
"Pedal torque adc step adv" is also an estimated value, for a real value a calibration with a weight is needed.

With torque sensor values ​​160 and 450 you have a delta of 290, much higher than TSDZ2.
It is an improvement, I do not see any problems.
Just consider increasing the maximum valid value for E02, now 500 is too close to 450.

mstrens, 2025-03-13:
The info about jerkiness in cadence mode came also from ebikestuff.
First they tested in hybrid mode and reported it. Afterwads they tested other mode and they noticed it also (so also in cadence where the torque sensor is not used).