emmebrusa/TSDZ2-Smart-EBike-1 #157: Question about the frames sent to the displays

mstrens, 2025-03-14:
I see that TSDZ2 motor is sell with original firmware with different displays e.g. VLCD5, VLCD6, 850C, 860C, XH18, B02N.

I presume that 
- TSDZ2 original firmware does not varies with the display provided by the seller.
- the original firmware depends only on the selected battery voltage and power). Here I do not take into account the V2 controller.
- the original firmware generates only one type of frame in the messages sent to the display.

  
Still I noticed that in your OSF version, even if all frames have the same length there are small differences. E.g. you have:
`
		// reserved for VLCD5, torque sensor value TE and TE1
		//#if ENABLE_VLCD5
		if ( m_config.enable_vlcd5) {
			ui8_tx_buffer[3] = (uint8_t)ui16_adc_pedal_torque_offset_init;
			if (ui16_adc_pedal_torque > ui16_adc_pedal_torque_offset_init) {
				ui8_tx_buffer[4] = ui16_adc_pedal_torque - ui16_adc_pedal_torque_offset_init;
			}
			else {
				ui8_tx_buffer[4] = 0;
			}
		} else if (m_config.enablec850){ //} #elif ENABLE_850C
			ui8_tx_buffer[3] = ui8_battery_current_filtered_x10;
			// battery power filtered x 10 for display data
			ui16_battery_power_filtered_x10 = filter(ui16_battery_power_x10, ui16_battery_power_filtered_x10, 8);
			ui8_tx_buffer[4] = (uint8_t) (ui16_battery_power_filtered_x10 / 100);
		} else { //#else
			ui8_tx_buffer[3] = 0x46;
			ui8_tx_buffer[4] = 0x46;
		} //#endif

`

If there is really only one version of the original TSDZ2 firmware, does it mean that this firmware is able to detect automatically the type of display being used?
I it can't detect automacally the display, then
- the assumption I made that there is only one version of the original firmware for all display is wrong
 or
- the display does not show the right data when used with original firmware.

I ask this because I saw a message from ebikestuff saying that EK01 shows the power but it overestimates the value by about 30%

emmebrusa, 2025-03-16:
There is an official Tonsheng protocol.
There is no display recognition.
Usually the firmware on the motor is standard, but there are also customized versions on request of dealers.
In the packet that the motor sends to the display there are two bytes indicated as reserved and used for this purpose.
On VLCD5 they are used to display TE and TE1, it is a standard function but is not documented in the protocol.
On some 850C (NOT ON ALL), they were used to display current and power, the same on a version for XH18 from a Korean dealer.
There is also a reserved byte in the packet that the display sends to the motor, used only in the first VLCD5 versions to set the maximum current, I don't think it was used anymore later.

On displays it is more complicated, the historical displays already do not respect the protocol, VLCD5, VLCD6 and XH18 have only 4 levels instead of 5, VLCD6 and XH18 have 4 SOC bars instead of 6, XH18 does not support all the error codes expected from 1 to 9.
The displays that arrived later are almost all derived from Bafang versions and none respect the Tonsheng protocol 100%.

On displays there are then differences that do not concern the stock firmware, but are important for OSF.
An example, the error codes displayed in the speed field instead of a separate field
There are also other differences and, to date, not all are considered in OSF.
Evaluating them all is a long job and all the displays should be available.

Regarding the power display on EDK01 for TSDZ8 (it is being discussed [here](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/154)), they probably use the two reserved bytes.
It should be checked if with VLCD5 and TSDZ8, TE and TE1 are displayed correctly as on TSDZ2.

emmebrusa, 2025-03-16:
Did you see that @ruedbi confirmed the consistency of the displayed power?
If you choose VLCD5 in the configurator, the TE and TE1 values ​​are sent to the display.
If you choose 850C, the current and power values ​​are sent to the same bytes.
So if ebikestuff tried setting 850C and the power does not match, the ADC input has a different scale.