emmebrusa/TSDZ2-Smart-EBike-1 #161: question about acceleration/deceleration % in javaconfigurator

mstrens, 2025-03-26:
This is not an issue for TSDZ2 but more a question for OSF TSDZ8.

In javaconfigurator, the tooltip about acceleration % gives some recommended values (from 0% up to 45% depending on the motor and battery voltages).

I see in the code that those % are mapped with
ui8_duty_cycle_ramp_up_inverse_step_default = map_ui8((uint8_t) m_config.motor_acceleration,
				(uint8_t) 0,
				(uint8_t) 100,
				(uint8_t) PWM_DUTY_CYCLE_RAMP_UP_INVERSE_STEP_DEFAULT,
				(uint8_t) PWM_DUTY_CYCLE_RAMP_UP_INVERSE_STEP_MIN);
So ramp up inverse can vary from 194 (0%) up to 117 (45%).
This counter is used in the interrupt running at 19kHz (so 1 step = 52 usec) to increase 
This means that to increase duty cycle by 1step (from 256) it can take from 10msec (0%) up to 6ms(45%)
In the "worst" case, to increase by 255 steps (from 0  to 255), it can take about from 2,5 sec (0%) up to 1,6 sec (45%).

The ramp down follows the same principle (with different defaults/%)
Ramp up and down can also varies based on other parameters (assist mode, cadenc rpm).

I have 2 questions:
- why changing the recommended values based on motor and battery voltage?
- could improper values generate low frequency (e.g. 3Hz) current and duty cycle periodic fluctuations? I ask this because in TSDZ8, I get such fluctuations and I do not know if it could be solved changing those parameters. Those fluctuations seems to exist for TDSZ8 even when I use a specific assist mode where I assign fixed values to controller current target and to controller duty cycle target (in order to avoid potential impact of some sensors like torque or cadence). 
At this stage, I do not know what I can do to avoid those low frequency fluctuations. 
I can add that I do not notice significant high frequency current fluctuation over an electric rotation (e.g. I traced the current for 256 successive steps of the 19khz Irq). So I expect that the magnetic flux is correctly generated.
It is now several days that I search for a solution.

emmebrusa, 2025-03-28:
The acceleration ramp mapping and the recommended values ​​are a legacy of the 20-beta-1 version from which I started.
The basic values ​​suggested when the Vmotor / Vbattery ratio changes, I think are intended to obtain the same acceleration, i.e. the same variation of revolutions in the same time.
This is because changing the voltage changes the maximum revolutions.
I have never verified the coherence of these values,
because no problems have ever been reported in this regard.
With a 36V motor and 36V battery I use the default value of 35%, but I have tried in a range from 5% to 60% without encountering problems.
Containing the acceleration ramp is to safeguard the mechanics and contain consumption.

The deceleration ramp mapping was not in the 20-beta-1 version, it was a constant value, with a rather long stopping time.
The first work I did on OSF, following the requests of those who wanted a faster stop, was to decrease the stopping time.
But then there were complaints in the opposite direction and I also understood that a too fast stop is not good, for this reason I added the deceleration ramp mapping.
Setting the deceleration ramp to 100% leads to a swing in the assistance, because it follows exactly the push on the pedals, which is an alternating movement.
This swing is attenuated by a longer deceleration ramp (lower percentage).

Regarding the fluctuation at the 3Hz frequency I do not know what to think.

mstrens, 2025-03-28:
Thanks again for the feedback.

I have perhaps an explanation for the 3 Hz frequency I got. I was using a home trainer with roller and an old bike with a very bad wheel (warped and off-center wheel). The 3 Hz seems to be the wheel frequency during the test. Probably that indeed the torque to be produced by the motor varies and so also the current.

I would like to come back on the FOC multiplier. I do not say that your experimentations where not exact and you do not get good results (at least for TSDZ2) but there are some aspects I really do not understand.
I presume you used experimentation based on the temperature to get the best motor efficency.  

In a previous issue (#issues 127), you explained the history of foc multiplier.
You provided this link 
https://www.sciencedirect.com/science/article/pii/S1876610211045309?ref=pdf_download&fr=RR-2&rr=91033c0a697f74ec

When I apply the formula for TSDZ2 48V and 5A (=250W) I get a quite different value:
Link says that foc angle should be 1/2 of and angle = L*I*ERPS/ U.
Imagine that we use the full power (so we should have about duty cycle = 100%, U = 48V and I = 5A) and that rpm = 3000. So eprs in rad would be 3000/60sec*8poles*6.28 = 2513.
So we have angle = 0.000135 * 5*2513/48 = 0,0368 Rad = 2,1 degree.
So the foc correction should be only 1/2 of 2 degree = 1 degree at about max power.
This is so far away of the value you use.

Another aspect I do not understand is that the formula (and original casainho code) takes care of erps in the calculation.
You do not take care of it.

Do you have any explanation?
Perhaps could mspider65 explain it.

emmebrusa, 2025-03-29:
The link I posted was just a hypothesis of the origin of the Casainho formula, I did not write that the formula gives reliable results.
The Casainho formula underestimates the angle by a lot, it is about 1/3 of the optimal one. This error was partially compensated with the offset = 10, which however with low currents was excessive.
The simplified formula of mspider65, takes into account the motor current, but also a delay in the combination of the hall sensors that changes with the speed of the motor.
For the details you should ask him, but if you look at the PWMCalculations.xlsx file you can understand something.
To find the optimal value of FOC_ANGLE_MULTIPLIER, mspider65 did it by checking the battery consumption on a reference climb.
I had tried it too but I had not obtained usable results.
With the home trainer I changed method, I obtained the optimal value by calculating the efficiency, Trainer power / Battery power.
This is the procedure I used:
- 860C has 9 levels, I use the levels as a step accelerator.
- The home trainer also has different load levels, I use 6 of them.
- I start the motor with the UP button, up to the test level, I set the load level on the home trainer.
At this point I turn on the lights to change the function of the UP and DOWN buttons, I use them to manually correct the foc angle, at level 5 no correction, higher levels positive correction up to +4, lower levels negative correction up to -4.
If the motor runs well I try to increase. Increasing the angle can have two consequences, increase in power and rpm and I can continue to increase, or power and rpm become unstable and I have to go back.
If instead the rotation of the motor is unstable already with the calculated foc (correction = 0), I try to decrease it until I obtain a regular rotation.
In this condition I calculate the efficiency, Trainer Power / Battery Power. Then I try to decrease the angle (-1 and -2) and check if the efficiency has changed.
I take note of the correction that gives a better efficiency.
- In these 54 combinations, I also take note of all the operating data.
Battery power and current with an external instrument.
Home trainer power with its app.
Duty cycle PWM, ERPS and FOC angle, read on the 860C display.
In addition to the correction of the foc angle already described.
I do not know the precision of the instruments used, but I consider the comparison valid.
- With this data I looked for a formula to obtain a foc angle that corresponds to those obtained manually in all 54 combinations, hence 50% battery current and 50% motor current, with consequent adjustment of FOC_ANGLE_MULTIPLIER.
- The tests were done with a 36V motor and a 36V battery, always fully charged. I don't have such a powerful stabilized power supply, so to test in the same conditions with medium-high power, I can only do two tests a day (I have two batteries), then I have to recharge them and continue another day.
You understand that the story becomes long, for this reason I have not yet done the tests with the stock display version.

mstrens, 2025-03-29:
Thanks for the detailed explanation.
I understand it was a huge work to make all those tests. Even if the formula seems strange, the most important is that it gives good results for TSDZ2. 

On my side I am always analysing why I get huge fluctuations of the current.
I already increased filtering on the ADC current  ( using the average of 2 measurements per Irq and using a smoothing filter where a new value count only for 1/8 of the filtered value).

The current fluctuates even with such filtering and when the irq used only a fixed duty cycle (because I request a fixed controller target current, a fixed controller duty cycle target and that there is no ramp up or ramp down (because measured current remains lower than current target). 

The average current over 1 or 2 sec seems quite stable but the min and max values are quite different (e.g max is regularly 3X greater than min when calculated over an interval of 1 or 2 sec).

Initially I suspected that it resulted from fluctuations of the torque resquested by the home trainer (using a very bad bicycle wheel).

Even this could play a role, it is not the only reason because when I calculate the min and max currents inside a single electric rotation, I also notice current fluctuations (but not so big). E.g. if min current is 0.8A, max current in the same electrical rotation can be 1A so 10% variation around avg.

I presume you did not noticed such current variations with TSDZ2?

Currently I am blocked and do not know anymore what I should try.

emmebrusa, 2025-03-30:
The fluctuations of the current could be caused by the inaccurate calculation of the rotor position.
I remind you that the optimization of the hall sensors was done on the TSDZ2. With the TSDZ8 it could be very different.
If you haven't already done it, take a look at this thread:
[https://endless-sphere.com/sphere/threads/optimized-tsdz2-firmware.110121/](https://endless-sphere.com/sphere/threads/optimized-tsdz2-firmware.110121/)

You should try to implement the calibration of the hall sensors, as mspider65 did in his android app.
These are complicated calculations with float, unthinkable to do them with STM8, with the TSDZ8 processor there should be no problems, but the code is written in java, it takes a bit of time to understand it.

The code can be found here:
[https://github.com/TSDZ2-ESP32/TSDZ2-Android](https://github.com/TSDZ2-ESP32/TSDZ2-Android)
and the file is this:
[https://github.com/TSDZ2-ESP32/TSDZ2-Android/blob/master/app/src/main/java/spider65/ebike/tsdz2_esp32/activities/HallCalibrationActivity.java](https://github.com/TSDZ2-ESP32/TSDZ2-Android/blob/master/app/src/main/java/spider65/ebike/tsdz2_esp32/activities/HallCalibrationActivity.java)

mstrens, 2025-04-02:
Thanks for the links.
I saw the thread of mspider explaining how he proceeded for the optimization. I followed another way (using an automatic update of the reference angles based also on the time intervals between hall pattern changes when the rotor runs at constant speed). 

Reading the thread of mspider, I understood that the total current (= battery current) should be constant (no spikes) when the hall sensors are well calibrated (and there is no automatic regulation in motor.c with ramp up/down e.g. because real current remains below the target and so duty cycle remains equal to the duty cycle target).
Still for TSDZ8 I get quite big current fluctuations.
Some fluctuations could be explained by the variations of the load (generated by the home trainer with an excentric wheel) and they are at low frequency(e.g. 1 or 3 Hz). When there is no load, those low frequency fluctuations does not exist.

What is very strange is that I get also current variations at a much higher frequency. In fact the frequency is 6 X the frequency of the electrical rotations (so equal the hall pattern change) and they have the form of a nice sinusoid.

E.g. when I set the controller duty_cycle on 80, the controller current tartget on 3A, I get
- with a load on the home trainer
- a fixed  g_duty_cycle of 80
- a real current that is about 0.6A average
- the interval between 2 rising edge of the same hall sensor (= a complete electrical rotation) is 18 msec 
- as the current is largely under the controller current target, it is normal that the g_duty_cycle doe snot vary. So this exclude external regulation.
- but the current varies from about 0,4 up to 0,8 A (measured by the MCU ADC)
- with the oscilloscope (connected to a shunt resistor inserted between battery and motor) I see cleary nice sinusoid every 3 msec

I expect that those sinusoid are not normal because:
- mspider says that current should be nearly constant
- the code in motor.c uses a regulation of the duty cycle (with ramp up/down) that is based on the current measured eacb 52 ns and not an average on a full electric rotation. So the regulation should not work very well.

Note:  I created a conversation on the forum named "Need help to fix an issue in OSF for TSDZ8" but I presume you did not see it.
There was a picture of the oscilloscope with the fluctuations (with no load on this picture).

Do you know how mspider proceeded to view/check the current (and spikes) on his oscilloscope.
I do not see another way than inserting a shunt resistor between motor and controller.

emmebrusa, 2025-04-04:
I saw the conversation, I didn't intervene because I don't know what to think.
Can you confirm that the oscilloscope image is the current between the controller and the battery?
Strange, it seems that the three phases are overlapped.
I tried on TSDZ2 with a shunt between the controller and the battery, the current is very disturbed, but mostly continuous.
I can't quantify the amplitude of the disturbances because I don't know the characteristics of the shunt.
If instead I try with a current transformer, I only see the disturbances, there is also a sinusoidal component, at a high and unstable frequency (tens of kHz), with an amplitude varying between 50 and 100 mA.
Test done with constant PWM and battery current almost 2 A.

In his tests for current measurements, mspider65 refers only to a multimeter with log recording and his android app.
Between the motor and the controller you can also use a current transformer.

dzid26, 2025-04-04:
It could be anything. 
- Have you validated voltage waveform shape e.g. in openloop with locked motor?
- Maybe your dead time is affecting the shape

mstrens, 2025-04-04:
Thanks for suggestions.

I do not understand what you mean with "voltage waveform shape e.g. in openloop with locked motor"
Which voltage should I measure? 
FYI, I looked at the voltage between 2 motor wires (adding a rc filter to filter the 19khz PWM) and I get nice sinusoid (at least as far I can see with an analog oscilloscope).

About the dead times, they are set on 1,5 usec. This seems a reasonable value. With no load on the motor, the consumed current is quite low (about 500ma on 36 V battery). The controller is does not become hot.  Original TSDZ8 firmware uses 1usec. I used 1,5 for safety. Could this explain a "nice" sinusoid on total current.

Note: I get a "nice" sinusoid when I optimise (manually in the software) the angles of the hall sensors. When I change the reference angles of hall sensors, the total current get much more distorsions.

dzid26, 2025-04-04:
> I looked at the voltage between 2 motor wires (adding a rc filter to filter the 19khz PWM) and I get nice sinusoid (at least as far I can see with an analog oscilloscope).

Sounds reasonable. 
The idea is if the sine waves are not ideal, their squares will not add up to a constant when calculating power. (As I understand, if power is not constant, then battery power and current aren't constant either).  
If those waveforms are non ideal sinewave, the oscillations will appear 6x in a period.  https://www.desmos.com/calculator/j2te3xeuvo

I mentioned deadband, because that could be a reason for sine imperfections. Another reason that comes to mind could be imperfect sine bemf. 

I don't know how accurate are waveforms on tsdz2 and whether the effect would be large enough to explain your current ripple. Just throwing ideas.

mstrens, 2025-04-08:
FY I tried to implement a real foc algorithm without using hall sensor (so sensorless) based on park and clarck transform.
I get also sinus wave sat the same frequency (6 X the erps). The amplitude of the sinusoid is similar.

So I expect this is the result of the befm (and not a bug in my original firmware).

I now plan to calculate the average current on a full electric rotation and use this average value for all regulations.