emmebrusa/TSDZ2-Smart-EBike-1 #164: Question about ui16_adc_motor_phase_current

mstrens, 2025-05-04:
In motor.c you have this test to limit duty cycle
(ui16_adc_motor_phase_current >  ui16_adc_motor_phase_current_max)

I ask myself if it makes really sense to check on motor phase current (with "motor phase current" being "battery current" / "duty_cycle".

It looks like when current is e.g. 10 A and duty cyccle is 50%, then phase current is 20A.

I do not understand the physical reason for such a test.
From my point of view, due to the motor inductance and due to the high frequency of the PWM signal, the real current inside the wires of the motor remain always quite close to the battery current (there are just small ripples).

If duty cycle is e.g. 50%, the current in the wires will at no time be twice the  "battery current".

I can understand that duty cycle has an impact on FOC angle but I do not see a reason to have protection on motor phase current.
Security checks on battery current seems me enough.

Perhaps I am wrong but then could you please explain.

dzid26, 2025-05-04:
> the real current inside the wires of the motor remain always quite close to the battery current (there are just small ripples).

No, the current will indeed be 2x battery current if duty is 50%. 
Only for 100% the battery current will be equal to the motor current. 

Energy dissipation in the motor is motor current squared so it makes sense to check it in principal.

emmebrusa, 2025-05-05:
@mstrens
The phase current limitation is mainly for the protection of the controller (mosfet).
It can attenuate the acceleration of the motor, but it is not a defect, on the contrary, it is useful when the user exaggerates with the "Motor acceleration" setting.
In acceleration the phase current could reach very high values.
The only doubt is that the limit value is too cautious.