emmebrusa/TSDZ2-Smart-EBike-1 #165: Not sure that torque sensor values are corrected when calibration and advance is used.

mstrens, 2025-05-25:
I do not fully understand the code used to calculate ui16_adc_pedal_torque_delta when calibration and adv are enabled.

I expect that the purposes are:
- to get a result (ui16_adc_pedal_torque_delta) in a range 0/160
- to let the user get a non linear relation between the raw ADC value and the result  (ui16_adc_pedal_torque_delta) in order to increase/decrease the sensitivity mainly at low value.

I expect OSF should provide a curve like show in the manual (for torque calibration) : https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/manuals/EN-Parameter_configurator_guide-TSDZ2-v20.1C.6-update.1.pdf

I tried to replicate the code in an XLS. Here the link to this sheet : https://docs.google.com/spreadsheets/d/1C25nZTsWT64aaJzKEXFSv45wIkMZjicUmnxiBm2NMbc/edit?usp=sharing

When I play a little with the parameters that the user is supposed to modify, I do not get the expected curve (e.g. when I change PEDAL_TORQUE_ADC_ANGLE_ADJ)

It could be that I made a mistake in the XLS but I do not find it (I checked it twice).

Please could you have a look at this xls.

emmebrusa, 2025-05-29:
The xls file you posted reproduces the code exactly, but the axes of the graph you see in the manual are delta and weight.
You probably didn't see the Torque_calibration.xlsx file in the manuals/xls folder.
Weight and Torque weight, I got them from measurements made on my first motor.
On other motors the profile of the curve may be different, but the logic doesn't change.

mstrens, 2025-05-29:
Thanks for having looked at my xls.
When I opened this issue, I did not realised that the values to be filled in the parameters used by the code are not the same values as those to be filled in the user interface (860c or javaconfigurator).
User manual says e.g. to use a value between 0 and 40 for the torque angle adj but this is in fact an index to an array and the values used in the code are then in range 160/16. 

I saw your file Torque_calibration.xls but it is not so clear how to used it and what is the purpose of the the 3 parameters with the suffix Adj (offset, range, angle). I expect yellow shows input cells but there are also some cells without formula that are not yellow. 

Are those 3 parameters foreseen 
- to let the user changes some behavior of the firmware (increrase/decrease sensitivity of torque sensor) without using torque_calibration.xls
-  or in order to the let the firmware align the curve (ADC <=> weight) of the torque sensor of a specific motor to a "target" (using the torque_calibration.xls) 

Is the xls to be used in this way: 
- measure the ADC torque sensor values for many different weights applied on the pedal (taking away the offset at no load).
- then fill those values in the column Torque weight
- then play with 3 parameters (offset adj, range adj and angle adj) in order to get one curve (which one : angle adj?) as close as possible of the "torque target" curve

Is it so that when those parameters are filled (+  Torque ADC offset + Torque ADC max) and flag calibrated is enabled, then the ui16_adc_pedal_torque_delta value calculated by OSF should be as close as possible of the "target" value?

I have also a question : for some assist mode (e.g. power assist), the human torque has to be calculated. Is it so that the human torque is just the ui16_adc_pedal_torque_delta  multiplied by a coefficient. Is it so that this coefficient has to be filled MANUALLY e.g. in 860c. The field to be used being Torque sensor ADC step (when calibration is disabled) or Torque sensor ADC step Adv (when calibration is enabled). The values in those parameters are not filled automatically but the user can let the 860C propose a value if he fill "Weight on pedal" and  "ADC torque step calc".

I expect that this coefficient can also be calculated in the Torque_calibration.xls and that the value is then in the cell named "STEP FOR POWER". Still the formula use also "PERCENT W WEIGHT". It is not clear how PERCENT W WEIGHT is found?

Sorry, perhaps all this is explained somewhere but I do not see where.

emmebrusa, 2025-06-06:
I saw your message but summer is coming, the time to dedicate to OSF is decreasing. In summer I take a break.

The version 20-beta-1 of @buba that I started with, only provided calibration with weight, you get the value of "Torque sensor ADC step" for the calculation of human power.
However, this method was little used for human power, because "Torque sensor ADC step" was used improperly to increase the assistance with a lazy torque sensor. It was the only parameter available, but it only worked in Power assist mode.
In the other assistance modes, "Torque sensor ADC step" is not used. To increase the assistance, only the level parameters were available, insufficient to correct the differences in torque delta, hence the need for a calibration to amplify torque delta.

An example to help you understand how much torque delta max can change on TSDZ2, the engine of my first bike had a delta of 160, I did 5000 km in the two years that I used it and it remained the same, the engine of the second bike had a delta of 80, after a year it dropped to 40, after two years it went back up to 60, after 5000 km it is stable at 60. They are two bikes that I now use little.

My first calibration only included two parameters
“Pedal torque ADC offset” and “Pedal torque ADC max”.
The same parameters were also used to correct offset and range. It was not very intuitive, furthermore the modification of one also affected the other, with the same offset value the effect was different depending on the range.
Consider that with a limited range, the offset correction (-5) was mandatory.
To overcome this, I added "Pedal torque ADC offset adjustment" and "Pedal torque ADC range adjustment".
"Pedal torque ADC angle adjustment" was added instead to solve the problem of a hand-bike user with limited physical strength in the arms.
On these bikes a coaster brake motor is used and, with the assistance at the maximum that was necessary for him, at full speed it was not the arms that dragged the pedals but the opposite.
A fine adjustment of the assistance was therefore necessary in the first kg of push.
The solution adopted did not satisfy him completely, but it improved a lot.

With "Weight on pedal" an estimated value of "ADC torque step calc" is calculated, a value that with 860C must be manually entered in "Torque sensor ADC step" if "Calibrated" is disabled, or in "Torque sensor ADC step adv" if "Calibrated" is enabled.
"PERCENT W WEIGHT" was obtained experimentally on my first two motors.
With a weight of 24 kg "Torque adc on weight" is about 75% of "Torque adc max".
I introduced this estimated value to avoid the user weight calibration.

The purpose of Torque_calibration.xls is only to see how the torque sensor curve changes by changing the parameters:
"Pedal torque ADC offset adjustment"
"Pedal torque ADC range adjustment"
"Pedal torque ADC angle adjustment"
after having entered "RANGE WEIGHT" (“Pedal torque ADC max” - “Pedal torque ADC offset”).