emmebrusa/TSDZ2-Smart-EBike-1 #166: Torque sensor for TSDZ8 can provide ADV values that exceed 500 on some motors

mstrens, 2025-06-05:
A user reported that his motor (TSDZ8) provides torque sensor values that exceed 500.

In the TSDZ8 firmware, I increased the value (500) used to check the validity of ADC values.
Still ideally, this should be changed also in javaconfigurator and in 860C firmware because TSDZ8 OSF uses the ADC offset max value to remap the torque values provided by the sensor to a max of 160 (to get a more standardised internal process).

Please adapt the javaconfigurator and 860c firmware.

Ps :  did you saw my message/discussion/questions about torque sensor from one week ago? I did not saw a reply.

emmebrusa, 2025-06-06:
Ok for 860C, I didn't understand what you changed in JavaConfigurator.

mstrens, 2025-06-06:
In javaconfigurator, it seems that the user can enter a value that exceeds 500.
Still there is a tooltip (message when mouse goes over the field) that says that max value is 500.

I do not understand why you think that I changed JavaConfigurator.

I did not change javaconfigurator but I copied it in my own github repository because : 
- I tried to upgrade TSDZ8 to your latest update and so I had to take care of the 2 fields you added at the end.
- if I take care of those 2 new fields in the HEX file, I also have to avoid that a user uses the new TSDZ8 version with an old config hex file
- So I had to use a new "main version" in the header.ini file

As I wanted to publish the code immediately, I copied the files related to javaconfigurator to my repository and I edited the file header.ini file. It could also that I will have to change a parameter like the foc multiplier when I get more feed back from user over the best value to use. 
So it seems me easier (at least at this stage) to have a copy of javaconfigurator files in my repository.

Note :  I think that you also reactivated a field about a flag for torque estimated. This field seems now to be used in latest version of OSF TSDZ2 (javaconfigurator generates a #define) but is not present in the generated hex file for TSDZ8. As this #define is in the middle of other existing #define, it is not so easy to add it at the same place in the hex file. It requires a synchronization between the javaconfigurator change and the OSF TSDZ8 change. If you want, you can add this field in the javaconfigurator and set main version on 6 (it is currently on 5 in my github and on 4 in your). If you inform me when it is done, I can then copy your new javaconfigurator.jar, update TSDZ8, and the header.ini file.

emmebrusa, 2025-06-09:
Ok, I'm on vacation at the moment.
I'll watch next week.

emmebrusa, 2025-06-16:
Here I am again.

Ok for ToolTipText update "Pedal torque adc max".
For TSDZ8, 650 is it ok?
Is it appropriate to indicate different values ​​also for maximum current and maximum power?

Yes, #define TORQUE_SENSOR_ESTIMATED was not used but was already present in the config.h file.
I did not check, I thought it was also present in the hex file.
If you think it is important to keep the same sequence I add it in the same position, otherwise I add it to the end of the file, as you prefer.

mstrens, 2025-06-16:
I hope you had good holidays.

650 seems me ok for ToolTipText "Pedal torque adc max"

It seems that TSDZ8 with original firmware supports 23A. So ToolTipText for "battery current max", could be changed.
About "maximum power" I think that there is currently only one tooltip: It is "street power limit".  ToolTip says 500W. TSDZ8 is announced for 750W but I presume it can deliver about 1000W in peak (23A *52V)

About Torque_sensor_estimated, I suggest to keep in HEX file the same sequence as in the config. Even if it is more work for me at this stage, it is less risky for future maintenance when I try to import your changes in the OSF TSDZ8 version. As said, it will just require a good coordination of TSDZ8 and javaconfigurator updates.

For you information, it seems that jerking issues for TSDZ8 have been solved. I noticed that ADC values can vary a lot with the position of the pedal. For some motors the difference between min and max values whith no load can be as low as 2 ADC steps but for others it can be about 30. So, I considered that I had to take care of the values over a whole pedal rotation.
I implemented 3 algorithms:
1- assistance based on the max ADC value between actual value, max of current 360° pedal rotation, max of previous 360° pedal rotation
2- assistance using the mspider algorithm (last 20 ADC strored in an array = 360 pedal rotation; assistance based usually on the average)
3- assistance based on a logic proposed by katana1234 (average of 40 last ADC; one ADC each 25 msec; when ADC differs a lot from previous, the latest value is added several times (up to 16X) in the queue to increase responsiveness).

It seems that better feeling is with 3, then 1. I was not satisfied by 2

emmebrusa, 2025-06-18:
Yes the holidays on the Finale Ligure trails went well. Thanks.

I'm happy to hear that the development of OSF for TDZ8 is going well and that jerking issues are being solved.

I updated Java Configurator, changes:
Added TORQUE_SENSOR_ESTIMATED in the hex file.
ToolTipText "Battery current max" 23 A for TSDZ8
ToolTipText "Pedal torque adc max" 650 for TSDZ8

Also updated configMainVersion,6 in the TSDZ8_header.ini file.

Let me know if everything is ok.