emmebrusa/TSDZ2-Smart-EBike-1 #167: Error TSDZ2-Smart-EBike-1-20.1C.6 with VLCD5

Bini-99, 2025-06-12:
I am delighted with the OSF software except for these errors:
After 15-30 minutes, the VLCD5 display switches continuously to 85.0 00 and xx
The battery display (max. 6 bars) no longer changes (always 4 bars) 
Press LED on > power is reduced.
A pause is inserted (Speed =00):  The VLCD5 displays correctly again; when driving on, Speed is displayed a few km 00 (compensation).
If Power (off/on) is switched, the display functions normally again. The battery display now shows the correct value (2 bars).
I would like the battery voltage to be displayed when switching on and then only as the original display.
As there are 5 bikes with a VLCD5 display in operation in my area, I do not want to use any other display.
Setting: Hybrid
Odometer:  compensation on 
Setting:  Sport

Bini-99, 2025-06-17:
I'm testing my bike with the TSDZ2-Smart-EBike-1-20.1C.6. with the setting: Hybrid, the VLCD5 switches to the display advanced settings (1) after about 15 minutes. 
The  functions don't work with the TSDZ2-Smart-EBike-1-20.1C.6-update-1 version of the VLCD5.

dzid26, 2025-06-17:
If you have trouble describing what is going on, take some photos and screenshots.   You need to put more effort into documenting the issue if you want to get help.

emmebrusa, 2025-06-20:
The test for random data display was to be done with v20.1C.6-update-1.
I already understand little of what you write, if other problems overlap, I understand even less.
I had asked to try v20.1C.6-update-1, because in version v20.1C.6 there were problems in managing data on the display, problems solved with version v20.1C.6-update-1.
However, I don't think it depends on that.
From what you wrote it seems like a communication problem between the display and the motor controller, check the wiring and connectors carefully. Also check the position/distance between the speed sensor and the magnet.
Between v20.1C.6 and v20.1C.6-update-1 nothing has changed regarding assistance/power, if you find a difference it is because there are different settings, check all the parameters.

Bini-99, 2025-06-21:
TSDZ2-Smart-EBike-1-20.1C.6-update-1  mit VLCD5
I cannot test the function with (update-1 & Default_Settings_TSDZ2_36V.ini) because the motor does not emit any power! The power of the motor is far too low!
The bike is unusable with these settings! Test it yourself!