emmebrusa/TSDZ2-Smart-EBike-1 #176: OverCurrent Check Delay Issues (False Positive)

Str4nger14, 2025-07-05:
- [ ] - **_Hi, I love emmabrusa firmware is fully customizable but i've an hard problem with setting up of the check delay of overcurrent delay, I've setting up the max check delay 125ms but for me is too low 'cause I receive false positive due too the battery 52v with motor 36v and current 14A with max 500W, so everytime I increase the gears if I put it too fast I receive Error07 and there're not ways to resolve it, only if I disable the entire function, so please, Can I setting up more than 125ms (5 in configurator) the check delay, 'cause I want this protection but I just want to increase it due too false positives issues, so the objective is to increase in some way maybe on the code or with an firmware update to increase it more than 5 Value, for the rest this software is amazing, THank you and have a nice day!_**

emmebrusa, 2025-07-06:
In my opinion, a value greater than 5 (125 ms) is equivalent to excluding it.
However, if you want to try to increase it, follow this procedure.
Open the Java configurator, set your preferences, click on "Compile & Flash".
Once the compilation is finished, when you see "Press any key to flash... (Ctl+C to stop)" exit by pressing Ctl+C.
Close the Java configurator.
With a text editor, open config.h
at the line #define "OVERCURRENT_DELAY 2", change the value to the one you prefer.
Compile & Flash without using the Java configurator, but with double clicking on compile_and_flash_20.bat

Str4nger14, 2025-07-06:
I understood, I set up a little bit due too exlude those false positives, thank you for your answer and help, really greatfull, have a nice day :D