emmebrusa/TSDZ2-Smart-EBike-1 #177: v20.1C.6-update-2 XH18 startup data values not showing

fusenuk, 2025-07-12:
I have an XH18 display and since flashing to v20.1C.6-update-2 it is completely ignoring the settings for showing data at startup and showing data when the light button is pressed. Turning the motor on just shows 0 for both scenarios.

I have flashed back to the previous version I was using - v20.1C.6-beta-1 - and this does show the data correctly but when you first turn the motor on it always displays '10' instead of the battery remaining % which is what I have set for the first data item to display. So at power on it shows '10' then the voltage but when you press the power button it shows the correct battery remaining % and then voltage.

I've flashed back to v20.1C.6-update-2 just before writing this message and the problem persists, so it doesn't seem to be a 'bad flash' or similar.

Before the beta-1 version I used to have the correct data items displayed on start up and with the light button which was TSDZ2-v20.1C.2-update-2

So to clarify -

v20.1C.2-update-2 = correct data items displayed and functions as expected
v20.1C.6-beta-1 = '10' always displayed instead of battery % remaining at startup but then normal functionality when using the light button
v20.1C.6-update-2 = no data items being displayed at all either at start up or with the light button

I'm using a TSDZ2b 48v with XH18

fusenuk, 2025-07-18:
I'll go through it again, I purposely don't copy INI files between versions, I open the UI side by side from old version to new version and go through each setting to compare and re-enter old settings into the new UI window. This is to specifically try to get around problems with different versions not liking old INI files.

I'm fairly certain all of the appropriate fields were set correctly as per old versions which all worked and was going to post screenshots.

I'll re-flash both of these previous versions over the weekend and include screenshots and the generated INI files.

fusenuk, 2025-07-19:
[TSDZ2-v20.1C.6-update-2 data blanking.zip](https://github.com/user-attachments/files/21329084/TSDZ2-v20.1C.6-update-2.data.blanking.zip)

I've attached a zip containing 2 screenshots of the basic GUI section and the advanced section, the generated .INI file and the generated config.h file.

This is for version TSDZ2-v20.1C.6-update-2 with a XH18 and when the display is turned on, the speedo just remains at 0 the entire time. If I then click the lights button, after a delay, it does eventually show the voltage then returns to 0.

emmebrusa, 2025-07-25:
I checked the ini file, the only issue is that you set VLCD5 instead of XH18.
This doesn't affect the data display, but only the parameter settings.
If when you turn on the lights, it displays 0 for 5 seconds and then displays the voltage for another 5 seconds, your battery probably was low.
The first value is the remaining battery capacity (0%).
Have you tried with a charged battery?

fusenuk, 2025-07-25:
Ah sorry, it looks like when I was using the 'copy experimental settings to proven' button I'd left the config selector on your default config, so the shared INI file was what came by default with the download, instead of the config I'd actually just flashed which is different and can be seen in the screenshots and the config.h file I included, which has the proper display type set -

```
#define ENABLE_VLCD6 0
#define ENABLE_VLCD5 0
#define ENABLE_XH18 1
```

This isn't a low battery problem. The voltage displayed is 50v right now when I can get it to show the voltage.

Current scenario -

20.1C.6-beta-1 = It shows '10' when first starting up but doesn't show a SoC % but then shows the correct SoC % and voltage when pressing the light button - (69% SoC and 50v currently)

20.1C.6-update-2 = It shows 0 when starting up with no SoC or voltage, and then when you press the light button it shows 0 for a couple of seconds and then shows the correct voltage of 50

I charged the battery from when I first reported this as it was around 30-40% to around 80% and then rode the bike for an hour which is why the current SoC is around 69%

If you are happy to provide an INI file that you think should display the data items at startup successfully, I'm happy to flash that file and I'll do a video recording of the configurator GUI showing the default values without me editing them, and then the XH18 display starting up to try and get an example of the problem.

emmebrusa, 2025-07-26:
Well, if you send me an .ini file that isn't the one you tested, it becomes difficult to understand.
I didn't check the correspondence between the .ini file and the config.h file.
I simply flashed your .ini file on my bike and it works fine.
In your .ini file, "Units type" is set to km/h, not mph!

You should know that setting mph has some limitations; the maximum speed that can be displayed is 62.4.
So the data that can exceed this value is divided by 10; among these data is SOC%; 99.9 will be 09.9.
However, there's another problem: the minimum displayable value is 03.6, so the SOC% display range is very limited.
Indeed, in the latest version, setting "Units type" to "mph" always shows 00.0; I need to figure out why.

I recommend setting "Units type" to "alt mph" and setting the display to km/h.
The speed will be displayed in mph even if the label is km/h, but the maximum displayable value becomes 99.9.
You will only lose the trip history, which is converted to km.
New trips will be added in miles.

Another tip: when trying a new version, to avoid setting errors or forgetting anything, feel free to use the INI files from previous versions; they are always compatible.

fusenuk, 2025-07-26:
Thanks. 

I've got a bit of middle ground that I'm happy with now -

I replaced the unit type if statements from update-2 (https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/src/ebike_app.c#L3168)
with the if statements used by beta-1 (https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/v20.1C.6-beta-1/src/ebike_app.c#L3084)

Swapping those around in the ebike_app.c file from latest Update-2 release gives the following:

At startup - you see '10' then it shows the SoC for a very small amount of time
On light button press - it successfully shows the SoC followed by the voltage

It looks like the addition of the brackets within the calculation is changing the SoC value to something that isn't valid in some way

emmebrusa, 2025-07-27:
Okay, I see.
I added parentheses to prioritize the operations.
When the data value is divided by 10, they work fine.
When the data value is multiplied by 10, I'm in the wrong place!
Thanks for the tip.